rockowitz/ddcutil #239: Release 1.2.1 and1.2.0 missing build tools (see screenshot) since 1.1.0.

shurushetr, 2021-12-06:
Hi, not sure if this on purpose, not critical, I will try to compile with files from 1.1.0 tar.
Building fo Rasp. PI4, current release in buster repo available is 0.9.2 at least for me
[image]
.

shurushetr, 2021-12-06:
Looks like tar files from git don't have the tools but this [http://www.ddcutil.com/tarballs/ddcutil-1.2.1.tar.gz](http://www.ddcutil.com/tarballs/ddcutil-1.2.1.tar.gz) has.

closed 2021-12-06 by shurushetr

rockowitz, 2021-12-08:
I'm not sure what you mean by "tools", but I see that the config directory of autotools generated files is empty on github but is included in the website tarball.  The web site tarball is created by autotools command "make diff", and contains some generated files.  Either tarball works if the build process begins with command "autoreconf".  However, the divergences between these tarballs should be minimized and documented. Thanks for pointing out the situation.  Is the difference just a curiosity for you, or has it caused build problems?

shurushetr, 2021-12-08:
I figured it out, I guess my limited knowledge of git is in play here. I
was trying to build the latest version of the app by compiling from source,
and git release archive didn't have anything in build folder. I then went
on the website and there was a previous version of tarball (110 I think)
with build folder complete. So I just replaced version numbers in URL for
older tarball and got correct tarball from website.

rockowitz, 2021-12-08:
I should have written "make dist", not "make diff".  Also, the tarball link on page www.ddcutil.com/building has been corrected. Thanks again.