rockowitz/ddcutil #240: Unfortunately ddcutil commands are slow

danielturus, 2021-12-10:
Hello!

First and foremost, thank you for this incredible piece of software! I am very greatful that I can control my external monitor in Linux.

For some reason, it takes about 4-5 seconds for a command to be applied using ddcutil cli.
If I use ddcui, the response is better (less then a second).
On Windows, using a different software the response is instant (so I presume it's not a hardware limitation).

Is there anything I can do to speed up the execution time?

Thank you very much!

rockowitz, 2021-12-10:
Normally, when **ddcutil** starts, it examines each /dev/i2c device to see if it is associated with a monitor that supports DDC.  This is a slow process, but even so 4-5 seconds seems excessively long.   By contrast, **ddcui** examines the /dev/i2c devices once at startup, so individual operations do not incur the overhead.

Most of **ddcutil**'s elapsed time is spent in pauses mandated by the DDC spec.  In addition, I2C is an unreliable protocol, so if there are data errors **ddcutil** has to retry operations. 

The ***--stats*** option will show a summary of recoverable errors, and also the amount of time spent sleeping.  Option ***--ddc*** will report data errors in detail.

If you're not seeing data errors, try using option ***--sleep-multiplier***, which adjusts the length of time spent in mandated sleep.  For example, if ***--sleep-multipler .2*** were specified, and the DDC protocol specifies that a program wait 40ms between the time it writes a request to the terminal and attempts to read a response, then **ddcutil** would instead wait .2*40ms = 8ms.  I've seen monitors that work well with ***--sleep-multiplier*** values as low as .1.  On the other hand, monitors that exhibit DDC data errors can sometimes benefit from values greater than 1. 

As noted above, **ddcutil** normally examines each /dev/i2c bus each time it starts. The ***--bus*** option can be used to specify the /dev/i2c device, avoiding much of the examination phase. 

Finally, if you have many monitors attached, i.e. 3 or more, the ***--async*** option will cause some of the monitor checks to be done in parallel.

Hopefully, the above information will address your performance issues.  Should you think it useful to continue discussion of this issue, please include the output of **sudo ddcutil environment --verbose** as an attachment.

danielturus, 2021-12-14:
Thank you very much for your response! I tried with many commands but works the best with the `--bus` flag (the execution time improved significantly (to about 100ms - 200ms)).

I have one monitor, LG 27UL600-W (4k display) connected with a HDMI 2.0 cable to a new, pretty performant PC with the following specs:
CPU: Intel i5 11600k Rocket Lake
GPU: NVIDIA GeForce GT 1030
MB: Asus Prime Z590-A

A bit of a newbie question, is there a way to replicate what `ddcui` is doing, but using `ddcutil` and keyboard shortcuts? Basically  to examine the /dev/i2c devices once at startup and when I'm using a keyboard shortcut to skip the phases already done at startup?

Thank you for your help!

rockowitz, 2021-12-15:
As I requested above, should you with to continue discussion of this issue, "please include the output of **sudo ddcutil environment --verbose** as an attachment.".  

Once you've used the ***--bus*** option there's not much information left that can be calculated once, and then reused.  The one exception is the capabilities string, which is constant for a given monitor.

danielturus, 2021-12-17:
Hey! I've added the output of sudo ddcutil environment --verbose below:
<details>
<summary>Click here to see the output</summary>
<pre>Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

The following tests probe the runtime environment using multiple overlapping methods.
   Current time (local): 2021-12-17 16:10:43 EET
   Current time (UTC):   2021-12-17 14:10:43
   Seconds since boot:   295

*** Basic System Information ***

ddcutil version: 0.9.8

/proc/version:
   Linux version 5.11.0-41-generic (buildd@lgw01-amd64-005) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021

Architecture:     x86_64
Distributor id:   Ubuntu
Release:          20.04
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/vmlinuz-5.11.0-41-generic root=UUID=cfe4b7fc-4f35-4ab4-8fb3-35389e577317 ro quiet splash vt.handoff=7

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 9.2.1

Processor information as reported by lscpu:
   Architecture:                    x86_64
   CPU op-mode(s):                  32-bit, 64-bit
   Byte Order:                      Little Endian
   Address sizes:                   39 bits physical, 48 bits virtual
   CPU(s):                          12
   On-line CPU(s) list:             0-11
   Thread(s) per core:              2
   Core(s) per socket:              6
   Socket(s):                       1
   NUMA node(s):                    1
   Vendor ID:                       GenuineIntel
   CPU family:                      6
   Model:                           167
   Model name:                      11th Gen Intel(R) Core(TM) i5-11600K @ 3.90GHz
   Stepping:                        1
   CPU MHz:                         3900.000
   CPU max MHz:                     4900,0000
   CPU min MHz:                     800,0000
   BogoMIPS:                        7824.00
   L1d cache:                       288 KiB
   L1i cache:                       192 KiB
   L2 cache:                        3 MiB
   L3 cache:                        12 MiB
   NUMA node0 CPU(s):               0-11
   Vulnerability Itlb multihit:     Not affected
   Vulnerability L1tf:              Not affected
   Vulnerability Mds:               Not affected
   Vulnerability Meltdown:          Not affected
   Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
   Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
   Vulnerability Spectre v2:        Mitigation; Enhanced IBRS, IBPB conditional, RSB filling
   Vulnerability Srbds:             Not affected
   Vulnerability Tsx async abort:   Not affected
[... 1002 lines cut ...]
   /sys/bus/i2c device numbers:   0 1 2 3 4
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        true
   loadable_i2c_dev_exists:       false
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     root
   cur_uid:                       0
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
</pre>

</details>

rockowitz, 2021-12-18:
You are running ddcutil 0.9.8, which is quite old.  In particular, it does not have the cached capabilities feature which speeds up execution of **ddcutil capabilities**. 

You are using the proprietary Nvidia driver, but do not seem to have enabled the special Nvidia configuration parameters.  See [Special Nvidia Driver Settings](https://www.ddcutil.com/nvidia/).  It is possible that **ddcutil** is having to recover from I2C communication errors.  Is the ***--stats*** option reporting errors?

RaafatTurki, 2023-02-23:
> I've seen monitors that work well with _**--sleep-multiplier**_ values as low as .1

This was it for me, mine even worked with `--sleep-multiplier=0.025` but I think I'll stick to `0.1`

closed 2023-04-01 by rockowitz