rockowitz/ddcutil #242: Is static build supported?

daravi, 2021-12-25:
After configuring with --disable-shared and --enable-static, I don't see any .a file in the build directory. Is this intentional? Are static builds not supported? (if so can this be added to the documentation?)

rockowitz, 2021-12-26:
If configured with --enable-static, **make** creates file libddcutil.a in directory src/.libs.  However, **make install** does not install the static library, given that distributions generally prohibit their installation.

**make install** could be modified to install the static library, leaving it to each of the package creation programs to eliminate libddcutil.a. What's your use case for the static library?

daravi, 2021-12-26:
@rockowitz Thank you for the clarification. I am not sure if having a static build would be absolutely necessary or not. I am creating a [Conan](https://conan.io/index.html) recipe for this library, and I want to allow the user of the recipe to be able to build this as shared or static as needed (unless there is some limitation that makes static builds not work). I know that sometimes you prefer to build a single binary for your application so I am guessing in that case having libddcutil as static would be useful.

In any case I think I can just manually copy the `.a` file for the static build instead of relying on `make install` to do it for me. But I would say if someone already set `--enable-static` it would make sense that they would want to install the static library as well when calling `make install`. (feel free to close this ticket)

rockowitz, 2021-12-27:
File src/Makefile.am in branch 1.2.2-dev has been modified to not delete libddcutil.a when installing ddcutil. By default, configure sets --disable-static, so distributions will not include the static library.  As you point out, if someone other than a distribution packager is building libddcutil, there's no reason to prevent the creation of a static library.

As an aside, the reason that the ddcutil executable is built without using the shared library is that ddcutil and libddcutil are interfaces (command-line and API) built on top of the core ddcutil functionality.  Internally, command-line ddcutil does not use the API functions.  There's much more functionality in ddcutil than is surfaced in the API.  

I took a brief look at Conan, which I wasn't aware of.  I'll be interested to see how ddcutil is packaged.

daravi, 2021-12-28:
Ah ok good to know. If you are interested, here is my PR for the recipe: https://github.com/conan-io/conan-center-index/pull/8555

daravi, 2021-12-28:
By the way I had to patch the makefile to remove the `-werror` flag since I get errors with Clang 10.0.1 in C++17.

closed 2021-12-28 by daravi

rockowitz, 2021-12-28:
Can you point me to a listing of the build, or do you recall what the fatal warning was?

daravi, 2021-12-28:
@rockowitz Here are the errors:
```
  CC       i2c_sysfs.lo
../../source_subfolder/src/i2c/i2c_bus_core.c:891:43: error: use of logical '||' with constant operand [-Werror,-Wconstant-logical-operand]
      if ( !(businfo->flags & (I2C_BUS_EDP||I2C_BUS_LVDS)) )
                                          ^ ~~~~~~~~~~~~
../../source_subfolder/src/i2c/i2c_bus_core.c:891:43: note: use '|' for a bitwise operation
      if ( !(businfo->flags & (I2C_BUS_EDP||I2C_BUS_LVDS)) )
                                          ^~~~~~~~~~~~~~
                                          |
../../source_subfolder/src/i2c/i2c_bus_core.c:1080:8: error: unused variable 'busct' [-Werror,-Wunused-variable]
   int busct = i2c_buses->len;
       ^

In file included from ../../source_subfolder/src/app_sysenv/query_sysenv.c:54:
../../source_subfolder/src/app_sysenv/query_sysenv_access.h:10:9: error: 'SYSENV_ACCESS_H_' is used as a header guard here, followed by #define of a different macro [-Werror,-Wheader-guard]
#ifndef SYSENV_ACCESS_H_
        ^~~~~~~~~~~~~~~~
../../source_subfolder/src/app_sysenv/query_sysenv_access.h:11:9: note: 'QUERY_SYSENV_ACCESS_H_' is defined here; did you mean 'SYSENV_ACCESS_H_'?
#define QUERY_SYSENV_ACCESS_H_
        ^~~~~~~~~~~~~~~~~~~~~~
        SYSENV_ACCESS_H_
In file included from ../../source_subfolder/src/app_sysenv/query_sysenv_access.c:37:
../../source_subfolder/src/app_sysenv/query_sysenv_access.h:10:9: error: 'SYSENV_ACCESS_H_' is used as a header guard here, followed by #define of a different macro [-Werror,-Wheader-guard]
#ifndef SYSENV_ACCESS_H_
        ^~~~~~~~~~~~~~~~
../../source_subfolder/src/app_sysenv/query_sysenv_access.h:11:9: note: 'QUERY_SYSENV_ACCESS_H_' is defined here; did you mean 'SYSENV_ACCESS_H_'?
#define QUERY_SYSENV_ACCESS_H_
        ^~~~~~~~~~~~~~~~~~~~~~
        SYSENV_ACCESS_H_
../../source_subfolder/src/app_sysenv/query_sysenv_access.c:173:8: error: unused variable 'busct0' [-Werror,-Wunused-variable]
   int busct0 = i2c_device_count();   // simple count, no side effects, consider replacing with local code
       ^

../../source_subfolder/src/app_sysenv/query_sysenv_sysfs.c:286:15: error: unused variable 'dev_ids2' [-Werror,-Wunused-variable]
   Device_Ids dev_ids2 = read_device_ids2(sysfs_device_dir);
              ^
```