rockowitz/ddcutil #243: DDC communication failed (displayport) BenQ XL2546 w/ Nvidia 2080 TI

errantmind, 2022-01-11:
Hello, I'm trying to get DDC to work and am having an issue, see below for basic info, or here for [ddc interrogate](https://pastebin.com/umidfQSU)

I have two monitors hooked up, one BenQ XL2546 over DP, one BenQ XL2411 over HDMI. No adapters. I had this same setup on Windows and DDC worked. Switched to Arch and now it doesn't appear to work.

I tried setting [nvidia params](https://forums.developer.nvidia.com/t/gddccontrol-issues-with-nvidia-drivers-i2c-monitor-display-ddc-dp-hdmi-failing/30427/61) and it didn't have any effect.

Any ideas on what is wrong?

```
Invalid display
   I2C bus:  /dev/i2c-4
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-4/name:   NVIDIA i2c adapter 3 at 1:00.0
   EDID synopsis:
      Mfg id:               BNQ
      Model:                ZOWIE XL LCD
      Product code:         32600
      Serial number:        E1K03607SL0
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2019,  Week: 3
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 09 d1 58 7f 45 54 00 00   ..........X.ET..
      +0010   03 1d 01 04 a5 36 1e 78 3e 90 55 a7 55 53 a0 28   .....6.x>.U.US.(
      +0020   13 50 54 a5 6b 80 d1 c0 81 c0 81 00 81 80 a9 c0   .PT.k...........
      +0030   b3 00 81 bc 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 20 2f 21 00 00 1e 00 00 00 ff 00 45 31 4b   E. /!........E1K
      +0050   30 33 36 30 37 53 4c 30 0a 20 00 00 00 fd 00 32   03607SL0. .....2
      +0060   f0 1e ff 3c 00 0a 20 20 20 20 20 20 00 00 00 fc   ...<..      ....
      +0070   00 5a 4f 57 49 45 20 58 4c 20 4c 43 44 0a 01 6e   .ZOWIE XL LCD..n
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

Display 1
   I2C bus:  /dev/i2c-5
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-5/name:   NVIDIA i2c adapter 5 at 1:00.0
   EDID synopsis:
      Mfg id:               BNQ
      Model:                ZOWIE XL LCD
      Product code:         32562
      Serial number:        G7G02200SL0
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2016,  Week: 29
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 09 d1 32 7f 45 54 00 00   ..........2.ET..
      +0010   1d 1a 01 03 80 35 1e 78 2e 9d e1 a6 54 54 9f 26   .....5.x....TT.&
      +0020   0d 50 54 a5 6b 80 d1 c0 31 7c 45 68 45 7c 61 68   .PT.k...1|EhE|ah
      +0030   61 7c 95 3c 31 68 02 3a 80 18 71 38 2d 40 58 2c   a|.<1h.:..q8-@X,
      +0040   45 00 13 2a 21 00 00 1e 00 00 00 ff 00 47 37 47   E..*!........G7G
      +0050   30 32 32 30 30 53 4c 30 0a 20 00 00 00 fd 00 18   02200SL0. ......
      +0060   78 0f 87 11 00 0a 20 20 20 20 20 20 00 00 00 fc   x.....      ....
      +0070   00 5a 4f 57 49 45 20 58 4c 20 4c 43 44 0a 01 76   .ZOWIE XL LCD..v
   VCP version:         2.2
   Controller mfg:      Mstar
   Firmware version:    2.1
   Monitor returns DDC Null Response for unsupported features: false
```

rockowitz, 2022-01-11:
Your observation that the XL 2546 fails no matter whether it is on an HDMI or DP connector argues against almost any suggestion I would have.  Does the XL 2411 work with both?  It really is an I2C level problem.  The output of i2cdetect shows slave address x37 responsive for /dev/i2c-4 but not /dev/i2c-5. If it's not too much trouble I'd suggest temporarily replacing the proprietary Nvidia driver with nouveau. If that works, it's clearly a Nvidia driver problem. I wish I had better suggestions.

errantmind, 2022-01-12:
Thanks anyway, that helps. I'll do some tests and try to narrow it down.

lpreams, 2022-01-24:
I may be having the same or a related issue. I have two BenQ GW2480 monitors, as well as two AOC 24G2W1G5 monitors. I also have an AMD Radeon 5700 XT. The monitors are connected to my Arch desktop as well as to a Windows 10 laptop. 

My only use case is switching inputs, so that's all I've tested. `ddcutil -b $monitor_number setvcp 60 $input_number` is able to switch away from and back to my Arch desktop on the AOC monitors just fine, but the BenQ monitors can only be switched away from the Arch desktop. Once they are displaying something else, `ddcutil` commands respond with `DDC communication failed for monitor on bus /dev/i2c-9`.

The monitors are connected to Linux/Windows via a mixture of HDMI and DP.

A freeware program on Windows called ControlMyMonitor is able to switch all four monitors in both directions.

[`ddc interrogate`](https://github.com/rockowitz/ddcutil/files/7929429/ddc.interrogate.txt)

closed 2025-02-10 by rockowitz