rockowitz/ddcutil #244: lenovo dock not working

danjenson, 2022-01-12:
I have 3 external monitors connected to my lenovo docking station, 2 ASUS and 1 Dell. the Dell and 1 ASUS are connected by HDMI in the rear, and one connected by thunderbolt usb in the front. Curiously, the one connected in the front by USB works with ddcutil, however, the others do not, and are not even detected by ddcutil. Is there a workaround or some firmware I should install?

Specs:
ThinkPad P1 GEN 2 (running Void Linux, kernel 5.15.12_1)
Lenovo ThinkPad Thunderbolt 3 Workstation Dock Gen 2
2 ASUS VS278Q-P 27
1 Dell UltraSharp U3415W
Nvidia Quadro T2000 (nvidia 470.94 driver)

rockowitz, 2022-01-12:
Please submit the output of  **sudo ddcutil interrogate --verbose** as an attachment.

danjenson, 2022-01-12:
```
Setting output level very-verbose...
Setting maximum retries...
Forcing --stats...
Forcing --force-slave-address..
This command will take a while to run...

The following tests probe the runtime environment using multiple overlapping methods.
   Current time (local): 2022-01-12 14:51:13 PST
   Current time (UTC):   2022-01-12 22:51:13
   Seconds since boot:   4329

General Build Options:
   BUILD_SHARED_LIB:    Defined
   ENABLE_ENVCMDS:      Defined
   ENABLE_FAILSIM:      Not defined
   ENABLE_UDEV:         Defined
   USE_X11:             Defined
   USE_LIBDRM:          Defined
   USE_USB:             Defined
   WITH_ASAN:           Not defined

Private Build Options:
   TARGET_LINUX:        Defined
   TARGET_BSD:          Not defined
   INCLUDE_TESTCASES:   Not defined

*** Basic System Information ***

ddcutil version: 1.2.1

/proc/version:
   Linux version 5.15.12_1 (voidlinux@voidlinux) (gcc (GCC) 10.2.1 20201203, GNU ld (GNU Binutils) 2.35.1) #1 SMP Wed Dec 29 19:18:57 UTC 2021

Architecture:     x86_64
Distributor id:   VoidLinux
Release:          rolling
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/boot/vmlinuz-5.15.12_1 root=UUID=44e330aa-fcde-4f98-b5c7-b09d44c745b8 ro loglevel=4

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 10.2.1

Processor information as reported by lscpu:
   Architecture:                    x86_64
   CPU op-mode(s):                  32-bit, 64-bit
   Address sizes:                   39 bits physical, 48 bits virtual
   Byte Order:                      Little Endian
   CPU(s):                          16
   On-line CPU(s) list:             0-15
   Vendor ID:                       GenuineIntel
   Model name:                      Intel(R) Core(TM) i9-9880H CPU @ 2.30GHz
   CPU family:                      6
   Model:                           158
   Thread(s) per core:              2
   Core(s) per socket:              8
   Socket(s):                       1
   Stepping:                        13
   CPU max MHz:                     4800.0000
   CPU min MHz:                     800.0000
   BogoMIPS:                        4599.93
   Flags:                           fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust sgx bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp sgx_lc md_clear flush_l1d arch_capabilities
   Virtualization:                  VT-x
   L1d cache:                       256 KiB (8 instances)
   L1i cache:                       256 KiB (8 instances)
   L2 cache:                        2 MiB (8 instances)
[... 3322 lines cut ...]
Call Stats:
   Type                                     Count    Millisec  (      Nanosec)
   write calls       (IE_WRITE)               258        3019  (   3019982593)
   read calls        (IE_READ)                258        3123  (   3123906890)
   open file calls   (IE_OPEN)                  1           0  (         4956)
   close file calls  (IE_CLOSE)                 1           0  (         6885)
   other I/O calls   (IE_OTHER)                 1           0  (          723)
   Totals:                                    519        6143  (   6143902047)

Sleep Call Stats:
   Total sleep calls:                                     301
   Requested sleep time milliseconds :                  14805
   Actual sleep milliseconds (nanosec):                 14861  (  14861798949)

Elapsed milliseconds since last reset (nanosec):     20601  (  20601288618)
Total elapsed milliseconds (nanoseconds):               60030  (  60030092916)

Total non sleep system call time:       6143 milliseconds
Total sleep call time:                 14861 milliseconds
Elapsed time:                          60030 milliseconds

Skipping invalid display on bus /dev/i2c-10

Display scanning complete.
```

rockowitz, 2022-01-14:
It appears that the Nvidia driver does not implement I2C support for docking stations.  This is not surprising.  I [reported](https://gitlab.freedesktop.org/drm/intel/-/issues/37) the problem over 4 years ago for the i915 driver.  The problem turned out to be more general.  The 5.15 kernel finally has implemented a fix for the drm drivers implemented in the kernel (i915, amdgpu, nouveau).  At least as an experiment, I suggest you try the nouveau driver. Also, you may want to post your problem on a Nvidia forum.

danjenson, 2022-01-14:
ok, thank you! I'll close for now, as I'll likely stick with the nvidia driver..

closed 2022-01-14 by danjenson

tukusejssirs, 2022-01-15:
I’ve just found [`i2c-nvidia-gpu`](https://www.kernel.org/doc/html/latest/i2c/busses/i2c-nvidia-gpu.html) ([Phoronix](https://www.phoronix.com/scan.php?page=news_item&px=nvidia-i2c-gpu-driver-turing); [its source code in the kernel](https://github.com/torvalds/linux/blob/master/drivers/i2c/busses/i2c-nvidia-gpu.c)), but I was not able to make it work by simply running `modprobe i2c-nvidia-gpu` (although `lsmod` contains `i2c_nvidia_gpu         16384  0`).

`lspci -v` contains the following:

```bash
01:00.0 VGA compatible controller: NVIDIA Corporation TU117GLM [Quadro T1000 Mobile] (rev a1) (prog-if 00 [VGA controller])
        Subsystem: Lenovo Device 22b8
        Flags: bus master, fast devsel, latency 0, IRQ 170, IOMMU group 1
        Memory at ed000000 (32-bit, non-prefetchable) [size=16M]
        Memory at c0000000 (64-bit, prefetchable) [size=256M]
        Memory at d0000000 (64-bit, prefetchable) [size=32M]
        I/O ports at 3000 [size=128]
        Expansion ROM at ee080000 [virtual] [disabled] [size=512K]
        Capabilities: <access denied>
        Kernel driver in use: nvidia
        Kernel modules: nouveau, nvidia_drm, nvidia
```

If anyone can tell me if I tested `i2c-nvidia-gpu` correctly, I’d be grateful.