rockowitz/ddcutil #251: DDC communication failed, but only from Linux

Userblues, 2022-02-07:
I ignored this because I didn't need it, until now. What puzzles me is ddcutil works from Mac. Linux desktop is connected to HDMI1, Mac to HDMI2. Both Linux and Mac have HDMI outputs, no adapters of any kind. From Linux I get communication failed, from Mac it works. Here is interrogate, perhaps there is something wrong with my kernel?
http://asclinux.com:8000/tmp/i0

rockowitz, 2022-02-07:
I strongly suspect that your V7 monitor on /dev/i2c-0 has a marginal I2C implementation.  Responses to DDC/CI requests should start with a single x6e byte.  In your case there are two.  Have you tried swapping the Dell and V7 monitors to see what happens?  Though the proprietary Nvidia driver for the monitor on /dev/i2c-3 appears to be working fine, if you try swapping you might want to apply the [special settings for the Nvidia driver](https://www.ddcutil.com/nvidia/).

Try setting ***--sleep-multiplier 2*** to increase the delay between when **ddcutil** sends a request to the monitor and when it tries to read the reply. Slowing things down can help with marginal monitors. Also, increasing the DDC retries from 10 to 15: ***--maxtries "15,15,15"***.

Lastly, and probably most usefully, I have made a change in development branch 1.2.3-dev. This isn't the first time I've seen reports of double x6e at the start of a DDC response.  The change regards the extra byte as a quirk and discards it.  If this happens, you'll see a message at the terminal. 

Let me know how it goes.

Userblues, 2022-02-08:
Thank you for looking into this. nVidia driver is present in the system, but for CUDA only. It has no displays connected to it. It has two kernel modules loaded, I unloaded both, just in case. All three displays are connected to the Intel onboard graphics. Third is my TV set, which was powered off. Monitor on /dev/i2c-3 is connected over VGA and yes, ddcutil works fine with it. Out of curiosity I powered on my TV, it is connected to DVI output with DVI to HDMI adapter, even sound works over this adapter. It also fails DDC communication on  /dev/i2c-2, probably the TV set does not have it.
Trying out the latest from git when I get a chance.

Userblues, 2022-02-08:
I built from git, no change. I also plugged my Linux to HDMI2 input of monitor, the same input Mac is normally connected to. No change, DDC communication failed. Latest interrogate was run with --sleep-multiplier 2.
http://asclinux.com:8000/tmp/ddcutil_git

Userblues, 2022-02-08:
OK, I messed up some information in my initial post. What works from Mac is ddcctl. From Linux ddccontrol sometimes works ...

rockowitz, 2022-02-09:
Be careful using multiple DDC programs.  There can be conflict.  In ddcutil, they appear as EBUSY errors.  The solution is command line option ***--force-slave-address*** which tries to seize the /dev/i2c bus, potentially clobbering the other programs.