rockowitz/ddcutil #252: verification failed for feature 10

Edesem, 2022-02-23:
Trying to change the brightness of my main monitor.  Using `sudo ddcutil -d 2 setvcp 10 brightness_level` however it returns `Verification failed for feature 10`

Running the command `sudo ddcutil -d 1 setvcp 10 brightness_level` and it works fine, So idk....

rockowitz, 2022-02-23:
Please submit the output of **ddcutil interrogate** as an attachment.

Edesem, 2022-02-23:
[ddcutil_interrogate.txt](https://github.com/rockowitz/ddcutil/files/8122850/ddcutil_interrogate.txt)

rockowitz, 2022-02-23:
My understanding of your environment is as follows: 
- The video card is an MSI GeForce GTX 1060 with 1 DVI connector, 1 HDMI connector, and 3 DP connectors. 
- You are using Nvidia's proprietary video driver. 
- Display 1 is /dev/i2c-4.  The Acer B277 uses the HDMI connector.
- Display 2 is /dev/i2c-5.  The Acer XB271HB uses a DisplayPort connector. 
- You are not using the [special settings](http://www.ddcutil.com/nvidia/) sometimes required for the Nvidia driver

To determine: Does the monitor brightness change but verification fail, or is the verification correct that the brightness hasn't changed.

Suggestions: 
- If  the monitor changes brightness correctly but the verification fails, use option ***--noverify***.
- Apply the special Nvidia settings.
- Give the monitor more time to respond.  Use option ***--sleep-multiplier 2.0***
- Since the problem appears only on the DP connected monitor, try using the DVI connector instead.