rockowitz/ddcutil #255: Allow overriding data type (SNC, CNC) when getvcp

jianhaozhou, 2022-04-12:
On BenQ PD2700U, the 0xDC feature controls the picture mode. The feature actually uses both sl and sh bits.

When writing, it works correctly.
`ddcutil setvcp -d 1 dc 0x0413`

But when reading using getvcp, it gets interpreted as SNC and sh bits are omitted.
```
ddcutil getvcp -d 1 dc
VCP code 0xdc (Display Mode                  ): Invalid value (sl=0x13)
```

Allowing overriding the type to CNC when reading would be very helpful.

jianhaozhou, 2022-04-12:
Just realized I could get raw value with --verbose.

closed 2022-04-12 by jianhaozhou

rockowitz, 2022-04-12:
Can you describe how the SH and SL bytes are used on feature xDC?  It may be possible to at least provide partial support for features of type CNC in [User Defined Features](https://www.ddcutil.com/udf).