rockowitz/ddcutil #256: Changing external monitor brightness without sudo

anishmittal2020, 2022-04-22:
Currently, I change external monitor brightness as follows:

```bash
	ValueNew="${1}"
	feature=10
	mapfile -t MonitorA < <(sudo ddcutil detect | sed -r -n -e "/Display [0-9]+/{n;s/I2C bus(.*)/\1/p;}" | grep -o .$)
	IFS=$'\n'
	for Monitor in "${MonitorA[@]}"; do
		ddcutil --bus="${Monitor}" setvcp $feature "${ValueNew}" &> /dev/null &
	done
```

Is it possible to do this without sudo? I added `ALL ALL=(ALL) NOPASSWD: /usr/bin/ddcutil` in `/etc/sudoers`, but I still get the following error:

```
Open failed for /dev/i2c-1: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-2: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-3: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-4: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-5: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-6: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-7: errno=EACCES(13): Permission denied
```

## Any suggestions?

rockowitz, 2022-04-22:
See [I2C Device Permissions](https://www.ddcutil.com/i2c_permissions) on the [ddcutil web site](https://www.ddcutil.com).

anishmittal2020, 2022-04-24:
- I added my user (nikhil) to group 12c

```bash
$ getent group i2c
i2c:x:137:nikhil
```

- I copied `45-ddcutil-i2c.rules`  to `/etc/udev/rules.d/`

```bash
$ ls -la /etc/udev/rules.d/45-ddcutil-i2c.rules 
-rw-r--r-- 1 root 328 Apr 22 12:39 /etc/udev/rules.d/45-ddcutil-i2c.rules
```

- I restarted my system. But I am still required to use sudo. Could you please tell me if I did something wrong. I read the links mentioned by you thrice and checked all the steps. Many thanks!

closed 2022-04-24 by anishmittal2020

rockowitz, 2022-04-24:
Since you've closed the issue I gather you've solved the problem.  But 
in case not, run "ddcutil environment" and look at the suggestions at 
the end of the output.