rockowitz/ddcutil #259: Error reading capabilities on LG IPS226

laggykiller, 2022-05-06:
Output of `ddcutil detect --terse`
```
Display 1
   I2C bus:  /dev/i2c-13
   Monitor:             GSM:IPS226       :SerialNumber
```

Output of `ddcutil capabilities`
```
Unparsed capabilities string: O��hX����/7j`�����C[U��hX����/7j`�����C[{��hX����/7j`�����C[��hX����/7j`�����C['��hX����/7j`�����C[���hX����/7j`�����C[���hX����/7j`�����C[���hX����/7j`�����C[���hX����/7j`�����C[���
Errors parsing capabilities string:
   Nothing follows segment name at offset 238
Model: Not specified
MCCS version: Not specified
VCP Features:
```

Output of `ddcutil capabilities` as byte sequence (From python)
```
b"Unparsed capabilities string: O\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[U\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[{\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[\x01\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02['\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[\xcd\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[\xd3\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[\xf9\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[\x9f\xff\xff\x02\x01hX\xba\x1b\x01\xfd\x93\xb3\x83/7j`\x81\xcc\xc9\xe3\xe8C\x02[\xa5\xff\xff\x02\nErrors parsing capabilities string:\n   Nothing follows segment name at offset 238\nModel: Not specified\nMCCS version: Not specified\nVCP Features:\n"
```

Output of `sudo ddcutil interrogate --verbose`
[ddcutil-interrogate.txt](https://github.com/rockowitz/ddcutil/files/8641201/ddcutil-interrogate.txt)

laggykiller, 2022-05-07:
I tried to use [softMCCS](https://www.entechtaiwan.com/lib/softmccs.shtm) in windows, and it seems like it can get it successfully

[image]

rockowitz, 2022-05-07:
@laggykiller  Just want you to know that I'm not ignoring your issue. I had minor surgery on Thursday and my brain is mush.   Will get back to you when I'm functional.  

For now, try using option  **-disable-capabilities-cache**.

laggykiller, 2022-05-08:
Unfortunately using --disable-capabilities-cache does not change the output...

Take care dev

rockowitz, 2022-05-08:
I have made a number of changes on branch 1.2.3-dev to improve remote debugging.  Please build from that branch, then execute the following command: 
~~~
ddcutil capabilities --disable-capabilities-cache --trcfunc parse_capabilities --trcfunc app_get_capabilities_string --trcfunc ddc_get_capabilities_string  --trcfunc multi_part_read_with_retry
~~~

laggykiller, 2022-05-09:
Here is the log

[ddcutil-log.txt](https://github.com/rockowitz/ddcutil/files/8648553/ddcutil-log.txt)

rockowitz, 2022-05-11:
Tracing clearly shows that invalid gibberish is being returned from the monitor.  We could enable even lower level tracing, but I have no expectation that it would show anything different.  If the problem were at the I2C level, I would expect to see invalid characters interspersed in otherwise readable text.   The data being returned doesn't look like any character encoding I know of.

Looking at the interrogate output, it appears you're using the proprietary Nvidia driver without the [Special Nvidia Driver Settings](http://www.ddcutil.com/nvidia/).  That would be the next thing to try.  Also, if possible, try using the monitor on a system using a non-Nvidia driver.

Note that capabilities string is reported for informational purposes only.  It is often inaccurate, and ddcutil does not rely on it.

laggykiller, 2022-05-11:
Not just on my machine with nvidia display card, but also my machine with intel internal graphics card has the same issue:

[ddcutil-interrogate.txt](https://github.com/rockowitz/ddcutil/files/8668533/ddcutil-interrogate.txt)

rockowitz, 2022-05-11:
You mentioned earlier that  using --sleep-multiplier did not help.  I've assumed you've used a value somewhat greater than 1, such as 2.0. Other than that I don't know what to try without having the monitor on my test bench and sniffing the I2C lines.

What's befuddling is that softMCCS is somehow able to obtain a readable string.  One guess, and it's a weak one, is that there's an old undocumented or alternative protocol for returning the capabilities string that the monitor uses and softMCCS understands.   Another is that the Windows I2C drivers are just better than those in Linux.  This is a relatively old monitor, and LG seems to have a cavalier attitude towards DDC standards.   The string reported by softMCCS has no model or cmds segments, and there are no lists of valid values for NC features.  The value for mccs_ver is 0.01, which is invalid. (Feature xdf reports 2.1, which is valid.)

botkero, 2022-11-26:
offtopic
hello @laggykiller 
can you provide a download option for softMCCS? The site is down and I can't find an alternative site to download.

laggykiller, 2022-11-27:
@botkero waybackmachine got that archived

botkero, 2022-11-27:
> @botkero waybackmachine got that archived, search for entechtaiwan

I feel stupid now, but I have no success with waybackmaschine. https://archive.ph/www.entechtaiwan.com has a few snapshots of the page, but none leads to a download for softMCCS. Can you give a link?
_The actual site of entechtaiwan.com is not down, but any traces to softMCCS are removed. A bit strange._

laggykiller, 2022-11-28:
@botkero Here is the archive of softMCCS:  https://web.archive.org/web/20220810190258/https://www.entechtaiwan.com/files/sm_setup.exe