rockowitz/ddcutil #260: Feature 0x60 does not support verification

daher13, 2022-05-16:
I'm trying to change feature 60 (Input Source) to the second option:
   Feature: 60 (Input Source)
      Values:
         11: HDMI-1
         12: HDMI-2
Using the command: 
ddcutil -v -d 1 setvcp 60 12

But it is returning:
Feature 0x60 does not support verification
Am I using the correct command? How to fix that?

rockowitz, 2022-05-17:
Use the ***--noverify*** option.  The problem with trying to verify the changed value of feature x60 is that once the input device changes it's most likely that trying to read the current value from the original input device will fail.

daher13, 2022-05-17:
The monitor blinks, but don't changed the input to another.

rockowitz, 2022-05-17:
This is a different question.  Assuming the values for HDMI-1 and HDMI-2 conform to the MCCS specification and are x11 and x12 on your monitor, you need to write **ddcutil setvcp 60 x12***.   Without the "x", the decimal value 12 becomes x0c.  

By convention VCP feature codes are always referred to by the their hex value, so feature x60 can also be referred to as 60.  However, in all other situations, values can either be decimal or hexadecimal, and so require "x" to disambiguate.

daher13, 2022-05-17:
Don't works yet. I'm trying: 
ddcutil --noverify -v -d 1 setvcp 60 x0c

rockowitz, 2022-05-17:
HDMI-2 is 0x12 = 18.   x0c is the value for Component video 1

i.e. you want **ddcutil setvcp 60 x12** or **ddcutil setvcp x60 x12** 

For values, see **ddcutil vcpinfo x60 --verbose**

daher13, 2022-05-17:
Didn't worked too. The verbose command output is:
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

VCP code 60: Input Source
   Selects active video source
   MCCS versions: 2.0, 2.1, 3.0, 2.2
   MCCS specification groups: Miscellaneous
   ddcutil feature subsets: 
   Attributes (v2.0): Read Write, Non-Continuous (simple)
   Attributes (v2.1): Read Write, Non-Continuous (simple)
   Attributes (v3.0): Read Write, Table (normal)
   Attributes (v2.2): Read Write, Non-Continuous (simple)
   Simple NC values:
      0x01: VGA-1
      0x02: VGA-2
      0x03: DVI-1
      0x04: DVI-2
      0x05: Composite video 1
      0x06: Composite video 2
      0x07: S-Video-1
      0x08: S-Video-2
      0x09: Tuner-1
      0x0a: Tuner-2
      0x0b: Tuner-3
      0x0c: Component video (YPrPb/YCrCb) 1
      0x0d: Component video (YPrPb/YCrCb) 2
      0x0e: Component video (YPrPb/YCrCb) 3
      0x0f: DisplayPort-1
      0x10: DisplayPort-2
      0x11: HDMI-1
      0x12: HDMI-2

I tried to exec with DisplayPort values, but monitor input didn't changed

rockowitz, 2022-05-19:
I'm at a loss for what else to add.  The **vcpinfo** output is telling you what all the possible values are for feature x60, according to the MCCS (Monitor Control Command Set) spec.  Per the spec, the values for HDMI-1 and HDMI-2 are x11 and x12.  No monitor implements all the possible values, and unfortunately, there are monitors that don't adhere to the spec.  You'll just have to test to determine what the actual values are for your monitor.  The **capabilities** command, which interprets the capabilities string reported by your monitor, can be helpful here, but its output is sometimes incorrect.

daher13, 2022-05-20:
I understood. I think that the monitor don't accept this function. Thank you!

denilsonsa, 2024-02-27:
I think this issue can be closed.

closed 2025-02-11 by rockowitz