rockowitz/ddcutil #261: soft lockup on AMD Ryzen 7 PRO 4750U

AndreasBackx, 2022-05-20:
I started experiencing soft lockups on my machine running Fedora 36 on kernel `5.18.0-0.rc7` (5.17 gives me some wifi lockups in the past, though ddcutil seems to be the issue since I updated to Fedora 36 from 35).

This includes the system logs from right before shutdown, interrogate info, and usbenvironment info: https://gist.github.com/AndreasBackx/ab480f22fb4b8b8b62ea66659efa0570

I had/have a similar issue on a discrete RX 6800, and I think it led me to an issue on Freedesktop (or another) GitLab, though I cannot seem to find it anymore. :( Let me know if I can provide some more information.

rockowitz, 2022-05-20:
Hello Andreas, 

First, thank you for providing detailed documentation.  Some points: 

- 0.9.9 is a very old version of ddcutil.  Please build from branch 1.2.3-dev.  There have been numerous changes that might affect what you're seeing.
- Checking for USB connected monitors is now disabled by default, since such monitors have proven rare. (Option **--enable-usb** enables it.)  So **ddcutil usbenvironment** is no longer relevant.
- The **interrogate** command includes **environment --verbose** along with other checks such as **probe** for each monitor.  For our purposes, there is no need to execute **interrogate**.  **ddcutil environment --verbose** contains all the information that might be relevant to this problem.
- The RX 6800 problem was a bug in the amdgpu driver for AMD Navi devices, which hung the system if an I2C bus that was in fact an SMBUS device was probed.  See #194. A workaround was added to ddcutil, and  in any event I believe the driver has  long since have been fixed.
- There have been a few reports of occasional hangs.  Remote diagnosis is very difficult and I have not been able to replicate the problem.  Issuing ddcutil commands in rapid succession seems to be a trigger.  If this is a ddcutil problem, hopefully we can figure out what's happening.
- Looking at your Python script, a few things stand out.  First, it appears you're simultaneously executing ddcutil in multiple threads.  Copies of ddcutil in separate threads do not communicate, so there could be some interference.  In 0.9.9, the instances could conceivably clobber each other if **--force-slave-address** is specified in each.
- If you're repeatedly calling ddcutil, using option **--busno** will be far less intensive than using **--serial-number**.  The latter essentially performs a **detect** command to inventory the monitors to associate the serial number with a bus number.  For example, try issuing command **ddcutil getvcp 10 --stats** with using **--serial-number** and then **--bus**.  The latter makes a fraction of the system calls, sleeps far less than the former, and therefore is significantly faster.
- As of branch 1.2.3-dev, option **--force-slave-address** no longer has any effect.  The EBUSY errors that it addressed are no longer possible. For details, see file CHANGELOG.md in the source.

Bottom line: 
- Build from branch 1.2.3-dev.
- Use --busno.  If need be, use a single **ddcutil detect --brief** call at the start of the script to build a table of bus numbers.
- Try accessing the monitors sequentially rather than in parallel, as a test.
- Insert delays between individual ddcutil calls.

I will upgrade my Fedora instance to 36 - it's currently at 35.  I'll also try running your script on my 3 monitor system.

Regards,
Sanford

rockowitz, 2022-05-21:
@AndreasBackx  i'm upgrading from F35 to F36 and noticed a ddcutil regression.   The existing /usr/bin/ddcutil is version 1.2.2.  The F36 upgrade is downgrading ddcutil to version 0.9.9-5.fc36.  If this downgrade happened to you as well, it might explain why your problem surfaced with the F36 upgrade.

AndreasBackx, 2022-06-20:
Apologies for the late reply. What you're saying is true, I will close the issue for now and reopen in case the problem persists for the latest version. Using the RPM from your COPR now, thanks so much for making that available!

closed 2022-06-20 by AndreasBackx