rockowitz/ddcutil #262: ddcutil dev regression

michal-josef-spacek, 2022-06-02:
Hi,
I tested ddcutil on my Thinkpad X270 and there are different results
Version in Debian
```
$ /usr/bin/ddcutil --version
ddcutil 0.9.9
Built without support for AMD Display Library (AMD proprietary driver).
Built with support for USB connected displays.
Built without function failure simulation.
Built with libdrm services.

Copyright (C) 2015-2020 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

```
$ /usr/bin/ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           LGD
      Model:
      Serial number:
      Manufacture year: 2013
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.
```

Version from git https://github.com/rockowitz/ddcutil/commit/3f37366455114b3b94babf86b1c4aaf9bf3ed49f
```
$ ddcutil --version
ddcutil 1.2.3-dev
Built with support for USB connected displays.
Built without function failure simulation.
Built with libdrm services.

Copyright (C) 2015-2021 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

```
$ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-6
ddcutil: i2c_sysfs.c:697: one_drm_connector: Assertion `( (has_drm_dp_aux_subdir) && (has_i2c_subdir) ) || ( !(has_drm_dp_aux_subdir) && !(has_i2c_subdir) )' failed.
Abort
```

Seems as regression

rockowitz, 2022-06-03:
Thank you for the report. 

The assert() failure is in recently added code that probes /sys for i2c related information.  Unfortunately, different video drivers use sysfs differently - there's no spec, I have changed many locations where the contents of /sys is not as expected from assert() failures to messages that data  is not as expected.  

Please build from the current  1.2.3-dev branch, execute command **ddcutil environment --verbose --f2** and submit the output as an attachment of some sort.  (**--f2** is a utility option used to trigger temporary code.)

michal-josef-spacek, 2022-06-03:
@rockowitz Seems to be fixed.
```
/usr/bin/ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           LGD
      Model:
      Serial number:
      Manufacture year: 2013
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.
```
**ddcutil environment --verbose --f2**
[OUT.gz](https://github.com/rockowitz/ddcutil/files/8835970/OUT.gz)

closed 2022-06-07 by rockowitz