rockowitz/ddcutil #266: Daewoo monitor, brightness not working

tralph3, 2022-06-11:
I have a Daewoo DW-MON22. It is a cheap 1080p monitor, but it does seem to be detected and working with `ddcutil`. However, the one thing I wanted to do with this tool, which is lowering the brightness, doesn't work.

You see, this option doesn't with with the built-in menu either. The slider is there, but you can move it from 0 to 100 and it does absolutely nothing. The brightness is the exact same. This is problematic since it is quite bright by default, and I often encounter myself using it in low light situations.

The menu also has some options that are greyed out. It's kinda weird.

The rest of the capabilities seem to work with `ddcutil`. I didn't try them all, but the ones I tried, apart from brightness, do seem to function properly. I grinned when I managed to turn off the monitor from a console command, that was neat.

I'm using Arch Linux with the latest, stock, Linux kernel. I'll attach the output of `ddcutil interrogate` below.

https://pastebin.com/axrUq73E

rockowitz, 2022-06-11:
Looking at the interrogate output, nothing jumps out at me.  However, I'll note that monitors often have modes such as movie or HDR in which certain features such as brightness are disabled.  I suggest you look around in the OSD for some such setting.  You might also try resetting to factory defaults.

tralph3, 2022-06-12:
This one doesn't seem to have any such modes. I have a slider for the backlight and another one for the brightness, neither do anything. Actually, the one for the brightness is currently set at 48, yet ddcutil reports it at 0. I can usin ddcutil to set it to other values, and doing gets after that does return the value I set, but the actual brightness of the monitor is unchanged. The OSD also doesn't reflect the change.

There's also tons more features that I completely forgot were disabled. For instance, I have some options to shift the image horizontally and vertically. In this menu, there's also options to auto detect the image position, but I can't select it, it jumps over it. Same with so many other settings.

My guess is that the same OSD is used for various monitor models, and the options not supported in certain models simply get disabled, and you can't select them. This doesn't explain why I can select and change both the backlight and the brightness, to no avail.

I don't want to reset to factory defaults, since getting a good configuration with this monitor was hell, something I don't want to go through again. Getting good colors was though, but I managed.

I don't know if any of this can be fixed, it may simply be that the monitor doesn't allow for these settings to be changed, and in that case ddcutil can't do anything about it.

I don't really have experience with debugging these things, but if there's any way I can help do ask. If there's no other avenues to try tho, we can just discard it as a problem of the monitor itself (being cheap and trashy), and not something ddcutil should worry about. In such case we could just close the issue and forget about it.

rockowitz, 2022-06-16:
Other than suggesting that you look for OSD modes, I'm afraid there's not much to be done.  Even name brand monitors can have a DDC/CI interface that does not correspond to the OSD.  See, for example, page [Notes on Specific Monitors](https://www.ddcutil.com/monitor_notes) which documents many ways in which the DDC/CI interface and OSD do not work as expected.