rockowitz/ddcutil #269: [Question] Is it possible to setvcp while your device is outputting to a different display?

William-Moriarty, 2022-07-07:
Hi,
I have a laptop connected to an ASUS monitor through a HDMI port. When I input `ddcutil detect` it correctly detects the ASUS monitor. When I set `ddcutil setvcp 0x60 0x01` it correctly switches the ASUS monitor to VGA output (which displays my W10 device). At this point I can still use my laptop using its native keyboard and screen (while the ASUS monitor displays whatever is on w10 device). The HDMI also remains physically connected.

When I do  `ddcutil detect` on my laptop at this point it tells me that I have an invalid display and the ddc communication failed.

```
Invalid display
   I2C bus:  /dev/i2c-1
   EDID synopsis:
      Mfg id:               ACI
      Model:                VS248
      Product code:         9368
      Serial number:        H7LMQS133650
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2017,  Week: 30
   DDC communication failed
```
Any `setvcp` or `ddcutil capabilities` command will fail, informing me  `Display not found`.

So based off this, my question is :
is it possible for my laptop to still use ddcutil to switch monitor source _while_ a different output on the external ASUS monitor is selected? Is there a workaround to this or is it a limitation? 

Thanks.

rockowitz, 2022-07-08:
It's a monitor limitation.  Some monitors accept DDC commands over any input.  Others accept DDC commands only on the current input.  Your monitor appears to be among the latter. To switch the input from your Windows system to the Linux system you'll have to use a Windows program like ClickMonDDC.

William-Moriarty, 2022-07-08:
I see, thank you. Is there a feature or a way to identify this limitation (in the case that someone may want to know if getting a new monitor)?

rockowitz, 2022-07-08:
I'm afraid not.  It's a behaviour not addressed by the specs, which monitor makers can implement one way or another. I doubt that a monitor maker's tech support would even understand the question, let alone know the answer.

I've added an entry for your monitor on page [Notes on Specific Monitors](https://www.ddcutil.com/monitor_notes).

William-Moriarty, 2022-07-08:
Thanks, much appreciation for your work on this project.

closed 2022-07-08 by William-Moriarty