rockowitz/ddcutil #270: DDC/CI does not work on secondary GPU (Both AMD)

gel-crabs, 2022-07-09:
I have a Ryzen 5 2400G APU driving 2 monitors and an RX 550 driving another monitor. Everything works fine on the RX 550, but the APU monitors don't work despite both having DDC/CI enabled.

Output of sudo ddcutil interrogate --verbose:
https://pastebin.com/TULeidjk

Output of ls -ld /sys/class/drm/card*/*/i2c*:
https://pastebin.com/c0ssD4dn

(Note: card0 is RX 550, card1 is Ryzen 5 2400G)

rockowitz, 2022-07-09:
Interesting problem you've got there.  

What I can see is that when interrogate tries to send a simple DDC/CI request to read brightness (feature x10) what comes back is a packet with all bytes zero. It might mean that the monitor is returning all zeros, more generally it indicates a DDC communication error of some sort. 

Have you tried any hardware permutations?  e.g. swapping monitors, running without the RX550? 

You appear to be running Wayland.  Have you tried just X11?

rockowitz, 2022-09-01:
@gel-crabs  FWIW, I finally freed up a motherboard with an AMD A6 APU for use in testing. The system has a Radeon Rx460 video card.  **ddcutil** detects all monitors (3 on the video card, 1 on the APU) both on Xorg and Wayland. I agree that the problem you're having is specific to your system.  If you come up with a solution, or at least more fully diagnose the problem, please post what you've found.

gel-crabs, 2022-09-04:
Yeah, it's likely to do with either my monitors (ASUS VX248H and HP 2010i) or my motherboard (ASRock B450M Steel Legend).

gel-crabs, 2022-09-04:
I should also mention that the ASRock B450M Steel Legend isn't particularily good with backlight support either, it errors without acpi_backlight=vendor.

closed 2022-09-04 by gel-crabs