rockowitz/ddcutil #272: Parseable Output

tylergets, 2022-08-08:
Love the project

I wanted throw up an issue to see if there is any interested in adding a parseable (maybe JSON?) output mode to ddcutil; This would make it easy for other applications to interface with it.

I would be happy to donate or contribute to this if theres interest and it'd be accepted into the project. Thanks!

rockowitz, 2022-08-10:
Take a look at ***--brief*** output for **detect** and **getvcp**.  It's intended to be machine readable. The **getvcp** output is the same form as is used by **dumpvcp**/**loadvcp**.  Does this do what you need?

That said, it would be cleaner if there were a standard format for parseable external files that was used by **dumpvcp**/**loadvcp** and user supplied feature definitions, and which could apply to **getvcp** and **detect**.  I've considered the enhancement, but the task is no where near the top of the to-do list.   I'm certainly open to a contribution that does this (perhaps using YAML rather than JSON) and would be happy to provide guidance.

mwoodson-coinbasecloud, 2022-09-16:
@rockowitz this project is really really cool!

However, I'd agree with @tylergets, it'd be really helpful to have parsable output.  JSON would be a standard options.

For example, it's really hard to parse this output:

```
# ddcutil detect --brief
Invalid display
   I2C bus:          /dev/i2c-1
   DRM connector:    card0-HDMI-A-1
   Monitor:          GSM:LG HDR WQHD+:203NTSUGW078

Display 1
   I2C bus:          /dev/i2c-2
   DRM connector:    card0-DP-1
   Monitor:          AOC:24G2W1G4:

```

Like, i want to associate the different Displays to Monitors.  This is hard to do with this format!

rockowitz, 2022-09-16:
As I've said, I think this enhancement makes sense.  I don't, however, see it rising to the top of my todo list any time soon.  The enhancement would entail modifying a well-contained portion of the large code base, and so is an ideal candidate for a user contribution.