rockowitz/ddcutil #273: ddcutil detect tries to open /sys/bus/i2c/devices/i2c--1

firmero, 2022-08-13:
Hi, when I've tried to use `ddcutil detect` I've got the following output:

```
Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Device /dev/i2c-255 does not exist. Error = ENOENT(2): No such file or directory
/sys/bus/i2c buses without /dev/i2c-N devices: /sys/bus/i2c/devices/i2c-255
Driver i2c_dev must be loaded or builtin
See https://www.ddcutil.com/kernel_module
Invalid display
   I2C bus:  /dev/i2c-5
   DRM connector:           card0-eDP-1
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:
      Product code:         4977  (0x1371)
      Serial number:
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2016,  Week: 11
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Phantom display
   I2C bus:  /dev/i2c-6
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL 275E2F
      Product code:         49722  (0xc23a)
      Serial number:        UHB2141009681
      Binary serial number: 9681 (0x000025d1)
      Manufacture year:     2021,  Week: 41
   DDC communication failed
   Use non-phantom device bus /dev/i2c-7

Display 1
   I2C bus:  /dev/i2c-7
   DRM connector:           card0-DP-2
   EDID synopsis:
      Mfg id:               PHL - Philips Consumer Electronics Company
      Model:                PHL 275E2F
      Product code:         49722  (0xc23a)
      Serial number:        UHB2141009681
      Binary serial number: 9681 (0x000025d1)
      Manufacture year:     2021,  Week: 41
   VCP version:         2.2
```
The output of `ls /sys/bus/i2c/devices` is :
```
1-0036  1-0037  1-0050  i2c-0  i2c-1  i2c-2  i2c-3  i2c-4  i2c-5  i2c-6  i2c-7  i2c-MSFT0001:00
```
I guess that i2c-MSFT0001:00 device is causing that issue in the function https://github.com/rockowitz/ddcutil/blob/b54f62a/src/i2c/i2c_sysfs.c#L1244 ( where i2c_name_to_busno returns -1 ).

julienfalque, 2022-08-14:
I'm using the [Gnome Brightness control using ddcutil extension](https://extensions.gnome.org/extension/2645/brightness-control-using-ddcutil/) which stopped working recently. In the logs I get a similar message:
```
ddcutil reading display status for bus: 4 is: Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Device /dev/i2c-255 does not exist. Error = ENOENT(2): No such file or directory
/sys/bus/i2c buses without /dev/i2c-N devices: /sys/bus/i2c/devices/i2c-255
Driver i2c_dev must be loaded or builtin
See https://www.ddcutil.com/kernel_module
VCP D6 ERR
```
The `i2c_dev` driver is loaded:
```console
$ lsmod | grep i2c_dev
i2c_dev                24576  0
```
Running the command to change the brightness does work:
```console
$ ddcutil setvcp 10 15
```

rockowitz, 2022-08-15:
@firmero Thank you for the precise bug report,.  Branch 1.4.0-dev contains a set of commits that address the problem.  Let me know if this works for you. 

It appears that you have a single external monitor that is detected at 2 different i2c addresses (a phantom display).  How is the display connected?  What is the laptop model?  Please execute **ddcutil environment --verbose** and  submit the output as an attachment of some sort.

rockowitz, 2022-08-15:
@julienfalque I was unaware of the Gnome Brightness Control.  If you can, please install branch 1.4.0-dev and let me know if it addresses the problem.

firmero, 2022-08-15:
@rockowitz thanks, changes resolved my issue. Ddcutil detect's output looks good to me: [ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/9338576/ddcutil-detect.txt).

Device `i2c-MSFT0001:00` is probably my touchpad (driver=i2c_hid_acpi). My setup contains a notebook display + external monitor connected with my ntb via the docking station (Dell Dock WD15). My environment: [ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/9338678/ddcutil-environment.txt)

Btw, Gnome Brightness Control extension stopped working, so I got here. That extension relies on one line output of `getvcp --brief D6 --bus $BUS` (then basic split on " " is used), but it was messed out with `Unable to open directory` error.

rockowitz, 2022-08-15:
@firmero  Thank you for the **environment** output. It helps me get a better understanding on how /sys is used for MST, so that I can more definitively discard the "phantom" display.  For a discussion of the problem, see issue [Same MST display appears as 2 /dev/i2c devices](https://gitlab.freedesktop.org/drm/intel/-/issues/3605#note_1409446)  on the drm/intel list.

julienfalque, 2022-09-08:
I just tried `1.4.0-dev` (d966330086caca93fdd71a8be43c77a903d1785f). It seems better but I think it's still not working as expected:
```console
$ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-3
   DRM connector:           card0-eDP-1
   EDID synopsis:
      Mfg id:               SHP - Sharp Corporation
      Model:                
      Product code:         5182  (0x143e)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2015,  Week: 40
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Display 1
   I2C bus:  /dev/i2c-4
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               GBT - UNK
      Model:                AORUS FI32U
      Product code:         12802  (0x3202)
      Serial number:        21400B000112
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2021,  Week: 40
   VCP version:         Detection failed
```
Changing the brightness still works but with error output and exit status:
```console
$ ddcutil setvcp 10 15
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
Setting value failed for feature 10, rc=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message
$ echo $?
1
```
Also, Gnome extension still does not work.

julienfalque, 2022-09-08:
Actually the behavior for `ddcutil setvcp 10 15` is the same on `1.3.0` with more errors, I forgot to report that in my first message:
```console
$ ddcutil setvcp 10 15
Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Device /dev/i2c-255 does not exist. Error = ENOENT(2): No such file or directory
/sys/bus/i2c buses without /dev/i2c-N devices: /sys/bus/i2c/devices/i2c-255
Driver i2c_dev must be loaded or builtin
See https://www.ddcutil.com/kernel_module
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
Setting value failed for feature 10, rc=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message
```

rockowitz, 2022-09-08:
@julienfalque Please run **ddcutil interrogate** (as root if necessary) and submit the output as an attachment.  Thank you.

julienfalque, 2022-09-09:
Here is the output:
* [with 1.3.0](https://github.com/rockowitz/ddcutil/files/9536883/ddcutil-1.3.0-interrogate.txt)
* [with 1.4.0-dev](https://github.com/rockowitz/ddcutil/files/9536884/ddcutil-1.4.0-dev-interrogate.txt) (d966330086caca93fdd71a8be43c77a903d1785f)

rockowitz, 2022-09-10:
The error your seeing is from reading the value after the write to verity that it has been set correctly.  You can disable verification using option ***--noverify***. 

More generally, there's something a bit marginal about your monitor, which can be seen in the retry data in the **interrogate** output.  Try using option ***--sleep-multiplier 2.0***.  Also if **ddcutil** commands are being issued in rapid succession from a script, putting a short sleep, say .5 seconds, between each **ddcutil** call can help.

As to the Gnome Brightness Extension, it's a "screen scraping" application that's not handling unexpected **ddcutil** output.  When I return from vacation I'll look into reducing the error messages in the **--brief** output that the Brightness extension uses, but a screen scraping application should  extension should be able to handle unexpected program output in some way.  

Finally, as noted previously, the error with double hyphens in /dev/i2c names, i.e. /dev/i2c--1", which exists in 1.3.0, has been fixed in 1.4.0-dev.   If it still occurs in in 1.4.0-dev that should be reported.

julienfalque, 2022-09-21:
I confirm increasing sleep multiplier fixes the remaining issue. Everything looks good to me, thank you!

liskin, 2022-10-03:
Worth noting that all this `Unable to open directory` output (all 5 lines of it) goes to **stdout** no matter that `--terse` or `--syslog` is specified. Hope this is addressed in 1.4.0-dev too.

fionnb, 2023-01-13:
I am getting this too.
After the latest Arch update to Version 1.4.0.r0.g670b6f36-1 ddcutil-git stopped working altogether (displays and features detected but all write attempts dit not change anything and then, of course, "Verification failed").
So I downgraded ddcutil  to 1.3.0 stable. Now value write changes _do_ work again now, but I get a lot of gibberish on every invocation:
```
Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Device /dev/i2c-255 does not exist. Error = ENOENT(2): No such file or directory
/sys/bus/i2c buses without /dev/i2c-N devices: /sys/bus/i2c/devices/i2c-255
Driver i2c_dev must be loaded or builtin
```
And yes, the driver module is loaded, of course :-)

rockowitz, 2023-01-13:
All the output you are seeing are the result of the mis-parsing of /dev/i2c devices with 2 hyphens in the name, which was fixed in release 1.4.0.

As a first step to diagnosing the problem, please try executing a setvcp command with option ***--trace all***.    Also, please run **ddcutil interrogate** so that I know your configuration.  And submit the output as attachments or some other method that is not in the text of your reply.  Thank you.

closed 2023-01-18 by rockowitz