rockowitz/ddcutil #274: ddcutil fails to read 2 byte vcp value

menkaur, 2022-08-27:
I'm using Samsung S34E790CN, and in this monitor, and I need to be able to change the input port. The problem is that in this monitor, input ports and dwords, and the display port I wanted is number 65289.

`sudo ddcutil getvcp 60 --display 2` produces

VCP code 0x60 (Input Source                  ): Invalid value (sl=0x00)

I'm assuming this is because the software only expects for the output to be 1 byte long, not 2 bytes, and it fails to process the first byte of information. Also, on this monitor, ddcutil capabilities does not seem to work, the output is:

Model: Not specified
MCCS version: Not specified
VCP Features:

Here's the output of ddcutil detect:

Display 2
   I2C bus:  /dev/i2c-3
   EDID synopsis:
      Mfg id:               SAM - Samsung Electric Company
      Model:                SE790C
      Product code:         3069  (0x0bfd)
      Serial number:
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2014,  Week: 39
   VCP version:         2.0

rockowitz, 2022-08-27:
@menkaur Please run **ddcutil interrogate** and submit the output as an attachment.  Thank you.

menkaur, 2022-08-28:
[interrogate.txt](https://github.com/rockowitz/ddcutil/files/9438879/interrogate.txt)
Here it is

rockowitz, 2022-08-28:
Next steps: 

1) Ensure you have enabled the [Special Nvidia Driver Settings](https://www.ddcutil.com/nvidia/)/ 

2) Execute this and the following commands, and again submit the output as attachment(s): **ddcutil getvcp 60 --bus 3 -verbose**

3) **ddcutil capabilities --disable-capabilities-cache --brief --bus 3 --trace all**

4) **ddcutil setvcp 60 65289 --bus 3 --trace all**

Also, do you have any idea what the high order byte in feature x60 is used for?