rockowitz/ddcutil #275: Suggestion - use uaccess udev rule so a group is not required.

digitaltrails, 2022-08-29:
Had you considered using a uaccess udev rule so that the i2c group is not required?

The rule I'm currently using for i2c is as follows:

```
cat > /etc/udev/rules.d/60-ddcutil.rules <<EOF
# local ddcutil udev rules
# User I2C/DDC/CI Access
KERNEL=="i2c-[0-99]*", TAG+="uaccess"
EOF
```
I accidentally found myself using such a rule because I had OpenRGB installed.  OpenRGB installs such rule for i2c, so I had never needed to setup any permissions to run ddcutil/vdu_controls. 

Using uaccess has the nice advantage that only the current graphical session owner has rw, and the permission is reassigned on login/logout. 

It would be easy to include such a rule file in the installation process (as with OpenRGB) because it doesn't required creating a group.

rockowitz, 2022-09-01:
Thank you for pointing out the **uaccess** tag.  I was not aware of it.

From initial search, it appears that uaccess is preferred over group assignment for systemd, but only systemd.   Brief testing shows that it does not work for ssh logins, but that is a minor use case. 

I will explore further and I don't identify any problems I'll document its use and likely install a  file using it in /etc/udev/rules.d. This would address the annoyance that **ddcutil** commonly works only from root after package installation.

rockowitz, 2022-10-24:
I have made a change in branch 1.4.0-dev that implements the uaccess change you suggested.  It's currently named 60-ddcutil.rules and is installed in /usr/lib/udev/rules.d or /usr/local/lib/udev/rules.d. 
The rule is : `SUBSYSTEM=="i2c-dev",KERNEL=="i2c-[0-9]*", ATTRS{class}=="0x030000", TAG+="uaccess" `

The ATTRS{class} test ensures that only /dev/i2c devices belonging to a video adapter are touched - ddcutil shouldn't be modifying permissions for devices that are irrelevant to it.  The SUBSYSTEM test seems redundant to me, but that's how the i2c-tools test is coded, so I'm following suit.

The environment command will need to be modified to take the change into account as well as documentation. 

Thanks again for pointing this aout.

rockowitz, 2023-01-18:
As of release 1.4.1, **ddcutil** installation installs file 60-ddcutil.rules in /usr/lib/udev/rules.d (or /usr/local/lib/udev/rules.d). The udev rule in this file uses tag **uaccess** to grant the logged on user RW permissions on any /dev/i2c device associated with a video adapter.

closed 2023-01-18 by rockowitz