rockowitz/ddcutil #276: Please add switching of preset monitor profiles (reading, cinema, etc.) for LG

hnet29a, 2022-09-03:
[interrogate.log](https://github.com/rockowitz/ddcutil/files/9482252/interrogate.log)
[getvcp_15.log](https://github.com/rockowitz/ddcutil/files/9482251/getvcp_15.log)
[setvcp_15.log](https://github.com/rockowitz/ddcutil/files/9482253/setvcp_15.log)

I switch profiles on the monitor and see how the value changes
in monitor - set profile reader
# ddcutil getvcp 15 --verbose
Feature definition file not found: GSM-LG_HDR_4K-30471.mccs
Getting data for non-table VCP code 0x15 - Unknown feature:
Raw value: opcode=0x15, mh=0x00, ml=0xff, sh=0x00, sl=0x01, max_val=255 (0x00ff), **cur_val=1** (0x0001)
VCP code 0x15 (Unknown feature               ): mh=0x00, ml=0xff, sh=0x00, sl=0x01
**cur_val=1** - this is profile reading

in monitor - set profile movie
# ddcutil getvcp 15 --verbose
Feature definition file not found: GSM-LG_HDR_4K-30471.mccs
Getting data for non-table VCP code 0x15 - Unknown feature:
Raw value: opcode=0x15, mh=0x00, ml=0xff, sh=0x00, sl=0x30, max_val=255 (0x00ff), **cur_val=48** (0x0030)
VCP code 0x15 (Unknown feature               ): mh=0x00, ml=0xff, sh=0x00, sl=0x30
**cur_val=48** - this is the movie profile

If i try to install
# ddcutil setvcp 15 48
Unrecognized VCP feature code: 0x15
Unrecognized VCP feature code: 0x15
# ddcutil setvcp 15 1
Unrecognized VCP feature code: 0x15
Unrecognized VCP feature code: 0x15

rockowitz, 2022-09-03:
**ddcutil** does not recognize feature x15 because it is not defined in the MCCS specification.  It is apparently an LG extension, but LG does not even list it in the capabilities string.  

You have two options: 
- Use option ***--force*** on the **setvcp** command
- Create a [feature definition file](http://www.ddcutil.com/udf/) for your monitor.

hnet29a, 2022-09-03:
> **ddcutil** does not recognize feature x15 because it is not defined in the MCCS specification. It is apparently an LG extension, but LG does not even list it in the capabilities string.
>
> You have [...]

thanks it works

closed 2022-09-03 by hnet29a