rockowitz/ddcutil #277: RFC: Merge and Collaborate on [MonitorControl/MonitorControl](MonitorControl)

zgover, 2022-09-04:
This would be a discussion post but the discussion capabilities are not enabled on this repository.

Interested to see if anybody would like to discuss the future life of this repository and whether or not it should be archived if all of the current contributors agree, so that efforts be transferred to the [MonitorControl/MonitorControl](MonitorControl) of which has much larger popularity for consumption and contributors. That way then we can provide two offerings, one being the API/CLI and the other being the GUI, ultimately also obtaining documentation of the feature controls for the solutions.

rockowitz, 2022-09-05:
MonitorControl certainly has a larger user and contributor base than ddcutil/ddcui, but I fail to see the synergy of merging them.  The former is Mac specific, the latter Linux (and potentially BSD) specific.  The effort and complexity of combining them would be significant.  Please be more detailed in the benefits you see to ddcutil/ddcui of bringing into MonitorControl, and conversely the benefits to MonitorControl. Also how you would manage the additional complexity.

zgover, 2022-09-10:
Merging them in the beginning will likely not benefit from unification of functionality and overall cross-platform portability. However the goal at least initially would be obtaining a greater success rate around discoverability and wide community support from consumer/contributor involvement. I searched for these exact tools about 4 years ago and never found anything that was great for Unix based environments, typically only Windows support for a miniscule amount of DDC controls that greatly lacked a wide area of coverage for monitor support. That being said, I didn't find this repo as quickly as I had hoped back when I was big into running Linux, however I found MonitorControl a few years ago and have search for alternatives a few times in the past and never came across this repo. The problem with both of these repos is they're slow to be discovered, yet most techy and non-techy individuals are craving support for such features, however most online resources and forums claim such capability are not possible for external monitors. 

So moral here is the greater we can obtain a wider contributor base supporting a monorepo of ddcutil and monitorcontrol, the greater chance both survive, and the sooner discussions may begin to consolidate into a single architecture to be built and ran for cross-platform support, and of course with your awesome level of documentation you've provided which MonitorControl lacks, let alone extensibility.

closed 2025-02-11 by rockowitz