rockowitz/ddcutil #280: ASUS PG329Q no serial number in EDID, best method of identifying specific identical monitors in dual/multi setup

bmcgruddy, 2022-09-13:
**TLDR** two identical monitors have no serial number, problem if trying to target in scripts.

Greetings,
Unsure if its a bug or (more likely) a quirk with this model of monitor specifically but the serial number in the EDID appears to be blank, my setup involves two identical [ASUS PG329Q](https://rog.asus.com/au/monitors/32-to-34-inches/rog-swift-pg329q-model/) which are detected within ddcutil.  

In a list of issues I'd say blank serial numbers is pretty minor one but this has come to a problem I've been unable solve, I wish to target one of them specifically to change its input source and since both of them report identically they have nothing unique between them to target. Currently I've been using the display or i2c bus number but that's only reliable if those numbers never change but obviously/alas that's not the case. 

Any help or advice on getting around this would be fantastic I've attached some relevant command outputs and the EDID dumps that I saved from nvidia-settings. 

Thank you for you time. 

[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/9554511/ddcutil-environment.txt) 
[ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/9554510/ddcutil-detect.txt)
[monitor-1-edid.txt](https://github.com/rockowitz/ddcutil/files/9555136/monitor-1-edid.txt)
[monitor-2-edid.txt](https://github.com/rockowitz/ddcutil/files/9555137/monitor-2-edid.txt)

rockowitz, 2022-09-14:
You've encountered an ugly little corner case.  It's mostly been seen on no-name monitors; I'm surprised to see it on one from Asus.  For this reason **ddcutil**,  unlike some other software, does not regard the 128 byte EDID as a unique identifier.   (For example, on the command line, the ***--edid*** option is treated as a search term.) 

Instead of using the I2C bus number directly, consider starting with /sys/bus/i2c/devices/i2-*/name, e.g. "NVIDIA i2c adapter 9 at 31:00.0",  to look up the bus number from the name.  The name is tied to the physical connector on the video adapter, and at least the first part, e.g. "NVIDIA i2c adapter 9", will be invariant unless you change the physical connector on the video adapter or the adapter itself.

bmcgruddy, 2022-09-16:
You're a legend, 
I was hopping that it was at least possible to do something like target the physical connector but I was unaware of the i2c device names I've already got something working based on your advice. 

Forgive me if I've totally misunderstood the field in question not that it seems to be currently an option for monitor selection anyway, but is "DRM connector" supposed to reflect the card/physical connector? 
I bring this up because ddcutil detect reports the same "card0-DP-2" for both monitors which had me a little puzzled as I was trying to find a workaround.

rockowitz, 2022-09-21:
I thought I had responded to your latest comment, but that's what happens when you're on vacation.  You encountered a bug.  **ddcutil** failed to get the get the connector for the display by traversing  /sys, so fell back to using the EDID, which in your case gave the wrong answer.  

It seems that the nvidia driver is using /sys in a way that I didn't expect.  Can you run **sudo ddcutil environment --verbose** and submit the output.  Thanks.

bmcgruddy, 2022-09-23:
[ddcutil-environment.txt](https://github.com/rockowitz/ddcutil/files/9631893/ddcutil-environment.txt)

No worries at all, hopefully you enjoyed your vacation to the fullest. :)

rockowitz, 2022-09-23:
The problem is that the the nvidia driver does not put enough information in /sys to associate an i2c bus number with a drm connector, so **ddcutil** falls back to using the EDID, which is ambiguous.  I'll look into at least providing a warning.

rockowitz, 2022-10-14:
@bmcgruddy Branch 1.4.0-dev contains changes to address the corner case where the connector name shown on **ddcutil detect --verbose** may be incorrect when two monitors have identical EDIDs and the proprietary nvidia driver is in use.  In that case, it is not possible to correctly determine the connector name. The command output includes a warning that connector names may be incorrect.

Please test **ddcutil detect --verbose** on your system.  If it fails to warn about the connector names, please issue the command **ddcutil detect --verbose --trcfile ddc_display_ref.c** and submit the output as an attachment.  Thank you.

bmcgruddy, 2022-10-25:
Apologies for the delayed reply, I had missed you message in my emails. I seem to coming across compilation errors in the 1.4.0-dev branch requested. (tested and was successful in compiling 1.3.2-release)
[log.txt](https://github.com/rockowitz/ddcutil/files/9857862/log.txt)

rockowitz, 2022-10-26:
I have modified source file i2c_edid.c to avoid the error. 

It's unclear to me what compiler option, or configuration option when building gcc, causes this variability. In addition to the tests I requested above, please submit the following:

Build **ddcutil** using command **make V=1**.  Extract the full gcc command used to build i2c_edid.c. 

Output of **gcc -v**

Thank you.

bmcgruddy, 2022-10-27:
No worries did a pull and attempted to re compile however still unsuccessful 
wish i was more than just a hobbyist python writer didn't even occur to me gcc/make would have verbosity levels.

```
git clean -fdx
git pull 
git checkout 1.4.0-dev
./autogen.sh |& tee -a log.txt
make V=1 |& tee -a log.txt
gcc -v |& tee -a log.txt
```
[log.txt](https://github.com/rockowitz/ddcutil/files/9874639/log.txt)

rockowitz, 2022-10-27:
I have updated branch 1.4.0-dev to avoid the compilation errors you hit.

More interestingly, the reason you hit the "a label can only be part of a statement" error and I do not is gcc version.  gcc 11 was changed to allow statements after labels after statements, gcc versions 10 and below show the error, versions 11 and above do not.  You are using gcc version 9, apparently because you're running Ubuntu 20.04

The EINVAL error was fixed by explicitly including file errno.h. Apparently it is not indirectly included in your environment. 

So now you should be able to test the changes to **ddcutil verbose -v**. 

Also, please execute **ddcutil environment --verbose** and submit the output as an attachment.  It will make diagnosing any additional problems easier.

bmcgruddy, 2022-10-28:
[log.txt](https://github.com/rockowitz/ddcutil/files/9885737/log.txt)
We have success :), I did need to include errno.h in i2c_strategy_dispatcher.c as well to get a successful compile. 
I can confirm that it gives a warning about the connector names too.
`Displays with I2C bus numbers 8, 9 have identical EDIDs.`
`DRM connector names may not be accurate.`

Trust the issues being related to still using the older LTS its about time I've refreshed my install. 
Thanks for your time sorry for the goose chase, let me know if you need me to test any other changes.

rockowitz, 2022-10-29:
First, I owe you an apology.  I hadn't already pushed the changes to github before commenting.

There's no need to apologize for a "wild goose chase".  **ddcutil** should build successfully using gcc 9.  One purpose of the published -dev branches is to catch obscure problems like the compilation error, and the interaction between duplicate EDIDs and the nvidia driver.  You've been very helpful identifying and diagnosing the problems.

closed 2023-02-02 by rockowitz