rockowitz/ddcutil #281: DDC communication failed ThinkVision T24d-10

anaegeli, 2022-09-14:
Hi
I have a problem with a T24d-10 connectect to a raspberry pi with hdmi.
You will find attached the ddcutil output.
What can I do to control the monitor.
What i need to be able to do is: 

- Turn the monitor on and off
- Set the brightness

[ddcutil_Debug.txt](https://github.com/rockowitz/ddcutil/files/9568263/ddcutil_Debug.txt)

Thanks for you're help.
Adrian

rockowitz, 2022-09-14:
It has been some time since I looked at the Raspberry PI, but here's what i observe/suggest. 

You have a DDC data error.  The video driver is reading all zeros from the monitor instead of a proper response to a DDC query. (Note: option ***--ddc*** can be used to report just DDC data errors.)

Have you tested with another monitor?

Try doubling the time that ddcutil waits between sending a request and reading the respone, using option ***--sleep-multiplier 2.0***.

Ir may be possible to slow the i2c bus speed using the dtparam baudrate option in config.txt.  I can't tell from config.txt what it's set to.  Try using 100000, which is standard speed, or if possible 50000, which is half standard speed.

I can't tell from the trace output what version of **ddcutil** you're running, but it's clearly not the It has been some time since I looked at the Raspberry PI, but here's what i observe/suggest. 

You have a DDC data error.  The video driver is reading all zeros from the monitor instead of a proper response to a DDC query. (Note: option ***--ddc*** can be used to report just DDC data errors.)

Have you tested with another monitor?

Try doubling the time that ddcutil waits between sending a request and reading the respone, using option ***--sleep-multiplier 2.0***.

Ir may be possible to slow the i2c bus speed using the dtparam baudrate option in config.txt.  I can't tell from config.txt what it's set to.  Try using 100000, which is standard speed, or if possible 50000, which is half standard speed.

I can't tell from the trace output what version of **ddcutil** you're running, but it's clearly not the latest version 1.3.0.  If you can, try building and using release 1.3.0 from github.

If any of this resolves your problem or at least leads to a solution, please post what you found so that others can learn from it.

If not, please execute **sudo ddcutil interrogate --verbose** and submit the output, again as an attachment.latest version 1.3.0.  If you can, try building and using release 1.3.0 from github.

If any of this resolves your problem or at least leads to a solution, please post what you found so that others can learn from it.

If not, please execute **sudo ddcutil interrogate --verbose** and submit the output, again as an attachment.

anaegeli, 2022-09-14:
I've tried multiple things, but couldn't update ddcutil to version 1.3.0

With the other commands, still nothing. 

Hier some output:
[ddcutil_Debug2.txt](https://github.com/rockowitz/ddcutil/files/9569802/ddcutil_Debug2.txt)

anaegeli, 2022-09-18:
Same device, same cable with added of HDMI to DVI Adapter, other monitor (HP E231) it works.
See Output File: (--version / interrogate --verbose / capabilities) [DebugE231.txt](https://github.com/rockowitz/ddcutil/files/9593735/DebugE231.txt)

My actual monitor has only: HDMI, DP or VGA. (ThinkVision T24d-10)
ddcutil is now on verison 1.3.0

anaegeli, 2022-09-18:
And here the Debug with ddcutil 1.3.0 (interrogate --verbose)
[DebugT24d.txt](https://github.com/rockowitz/ddcutil/files/9593755/DebugT24d.txt)

rockowitz, 2022-09-20:
Your results so far suggest that the problem is the monitor.  If you are able to test your monitor on a desktop system running **ddcutil** that would confirm it.

For our purposes, HDMI and DVI are interchangable, so using an adapter cable is not a problem. Both DVI and HDMI use the same video signaling protocol (TDMS) on the video pins, and use the same I2C protocol on the I2C related pins.  HDMI has some additional lines, such as CEC (Consumer Electronics Control), that don't matter in our context.