rockowitz/ddcutil #283: LG 34GK950-F Input switch to working.

DSDV, 2022-10-05:
Hello Dev & Community <3,

i would love to input switch via CLI / Hotkeys.
I tried to follow the https://www.ddcutil.com/ website.
Changing Brightness is working as expected. So far so good.

Sadly when i try to input switch there is no error message. But the monitor does not switch :(

[Monitor 1 Bus 11]

with kind regards

DSDV, 2022-10-05:
` sudo ddcutil interrogate                                                                                                                                                    1 ✘ 

[sudo] password for hq: 
Setting output level very-verbose...
Setting maximum retries...
Forcing --stats...
Forcing --disable-capabilities-cache
This command will take a while to run...

The following tests probe the runtime environment using multiple overlapping methods.
Set environment variable SYSENV_QUICK_TEST to skip some long-running tests.
   Current time (local): 2022-10-05 14:12:01 CEST
   Current time (UTC):   2022-10-05 12:12:01
   Seconds since boot:   19322

General Build Options:
   BUILD_SHARED_LIB:    Defined
   ENABLE_ENVCMDS:      Defined
   ENABLE_FAILSIM:      Not defined
   ENABLE_UDEV:         Defined
   USE_X11:             Defined
   USE_LIBDRM:          Defined
   USE_USB:             Defined
   WITH_ASAN:           Not defined

Private Build Options:
   TARGET_LINUX:        Defined
   TARGET_BSD:          Not defined
   INCLUDE_TESTCASES:   Not defined

*** Basic System Information ***

ddcutil version: 1.3.0

/proc/version:
   Linux version 5.15.65-1-MANJARO (builduser@fv-az183-516) (gcc (GCC) 12.2.0, GNU ld (GNU Binutils) 2.39.0) #1 SMP PREEMPT Mon Sep 5 10:15:47 UTC 2022

Architecture:     x86_64
Distributor id:   ManjaroLinux
Release:          22.0.0
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/@/boot/vmlinuz-5.15-x86_64 root=UUID=327bd87e-b844-4e25-9c10-5189f0298221 rw rootflags=subvol=@ quiet amd_iommu=on iommu=pt hugepages=8192 initcall_blacklist=i2c-designware-pci cryptdevice=UUID=019ae62b-2d6b-4cc9-ae83-a119fe458083:luks-019ae62b-2d6b-4cc9-ae83-a119fe458083 root=/dev/mapper/luks-019ae62b-2d6b-4cc9-ae83-a119fe458083 splash apparmor=1 security=apparmor resume=/dev/mapper/luks-28e7d411-2495-45f3-b4dd-47c94d9fc7e0 udev.log_priority=3

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 12.1.0

Processor information as reported by lscpu:
   Architecture:                    x86_64
   CPU op-mode(s):                  32-bit, 64-bit
   Address sizes:                   43 bits physical, 48 bits virtual
   Byte Order:                      Little Endian
   CPU(s):                          12
   On-line CPU(s) list:             0-11
   Vendor ID:                       AuthenticAMD
   BIOS Vendor ID:                  Advanced Micro Devices, Inc.
   Model name:                      AMD Ryzen 5 3600 6-Core Processor
   BIOS Model name:                 AMD Ryzen 5 3600 6-Core Processor               Unknown CPU @ 3.6GHz
   BIOS CPU family:                 107
   CPU family:                      23
   Model:                           113
   Thread(s) per core:              2
   Core(s) per socket:              6
   Socket(s):                       1
   Stepping:                        0
   Frequency boost:                 enabled
   CPU(s) scaling MHz:              55%
[... 3017 lines cut ...]
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values (unparsed): 05 08 0B 
      Values (  parsed):
         05: 6500 K
         08: 9300 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 52 (Active control)
   Feature: 60 (Input Source)
      Values (unparsed): 11 12 0F 10 
      Values (  parsed):
         11: HDMI-1
         12: HDMI-2
         0f: DisplayPort-1
         10: DisplayPort-2
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B2 (Flat panel sub-pixel layout)
   Feature: B6 (Display technology type)
   Feature: C0 (Display usage time)
   Feature: C6 (Application enable key)
   Feature: C8

DSDV, 2022-10-05:
thank you for your quick reply. 
1. thanks again for the clarification <3
2. ill try what you suggested and post again
3. i know that for windows there is a tool from lg to change monitor settings and i saw that ther are features like auto setup which are set to off - can they have an impact?

sorry for messing up the post. I did not find another button for posting code :(

rockowitz, 2022-10-07:
Re point 3, I would doubt that autosetup  affects input source, but the only way to tell is to test.  Usually anything that can be changed using the LG Windows monitor program can also be changed using the On Screen Display. If you can connect the monitor to a Windows system that would be instructive. Verify that you can change input source using the OSD.

Re attaching files, there indeed is no button but as I'm sure you've noticed by now at the bottom of the comment block is a section indicating you can "Attach files by dragging & droping, selecting or pasting them."  Or you could create a gist and include the URL in your post.

deodoro, 2022-11-25:
Hi there @DSDV, I hope my findings below may be of any use to you.

I use a couple of LG35 ultra-wide. Monitor 1 was connected through USB (HDMI adapter), monitor 2 through  native HDMI output (Mac Mini). Brightness, volume, etc all work fine for both. But it turns out while monitor 1 switches just fine, while 2 yields similar behavior to what you witnessed (99% of the time executes with no errors).

I plugged out the HDMI from the Mini and added another USB-C HDMI adapter. It works just fine now, I can switch back and forth HDMIs and Thunderbolt on all monitors and computers  with StreamDeck buttons. 

My hunch is that the root cause (at least for the problem I see) may be traced to the Mac HDMI firmware, neither DDCCTL nor LG.

closed 2023-02-02 by rockowitz