rockowitz/ddcutil #287: ABI forward compatibility

albertvaka, 2022-10-29:
I'm considering making libddcutil a default dependency for KDE's powerdevil.

It is important that KDE binaries don't break when a shared library is upgraded, which means the ABI of the libraries we depend on should not change (at least between minor versions).

I've seen that, at least between 1.2.0 and 1.2.2, some functions were removed. I'm not sure if these were only internal functions that would not be used from an external application, or if this would actually break external apps unless they are rebuilt. 

Is forward ABI stability a guarantee we can have from libddcutil?

rockowitz, 2022-10-29:
Starting with ddcutil 1.0.0, libddcutil numbering has followed the standard Linux convention, and is implemented using libtool.  The major version is incremented for non-upwardly compatible changes.  The minor version is incremented for API additions.  The micro/patch version is incremented for internal changes.

The API is specified in header files ddcutil_types.h, ddcutil_c_api.h, and ddcutil_status_codes.h.  The API has been stable for some time now, with only function additions.   Whenever possible, deprecated functions either become noops or are implemented using other functions; they will only be removed if the major version changes.  Adding enum values is treated as an upwardly compatible change, providing all existing values remain the same.

Having said all that, has the PowerDevil code been updated recently?  Last I checked, it was not suitable for production use.  See this [comment](https://github.com/rockowitz/ddcutil/issues/205#issuecomment-832397668) on an earlier issue thread from the person who implemented the PowerDevel/ddcutil code.

albertvaka, 2022-10-30:
The main concern from @d-vogel in that comment is:
> So there's a good chance the ddcutil API version used in powerdevil is now mostly deprecated.

I'm not sure if that's the case (maybe you can say if there's something terribly wrong by looking at [the code that uses libddcutil](https://invent.kde.org/plasma/powerdevil/-/blob/master/daemon/backends/upower/ddcutilbrightness.cpp)?).

However, although the current implementation is very simple (eg: we don't support having separate brightness sliders for each monitor) it seems to be working well for me and some other people that use it.

rockowitz, 2022-11-09:
I apologize for taking so long to answer your question.  I needed to refresh my understanding of PowerDevil. 

I looked at PowerDevil in late 2021 because of an assert() failure in libddcutil triggered by a call from PowerDevil after a return from screen sleep.  The program  occurred in ddca_open_display2()  when attempting to open a display that was already open. Upon return from screen sleep, PowerDevil was creating a new display reference and trying to open it, even though the existing handle had not been closed using ddca_close_display(). As of release 1.2.1, ddca_open_display2() have been fixed to instead return DDCRC_ALREADY_OPEN if the display is open in the current thread (in addition to DDCRC_LOCKED if the display is open in a different thread). 

It appears that PowerDevil has been modified to reuse the existing display handle on return from screen sleep instead of trying to create a new display handle, which is probably why you have not encountered a problem, though you need to check that PowerDevil does indeed not fail in this situation. 

One other thing I'd point out is that (for most features, including brightness) ddca_set_nontable_vcp_value() reads the value from the display after sending the message to the display to set it.   You my want to call ddca_enable_verify(false) to disable this behavior.

albertvaka, 2022-11-09:
Thanks a lot for looking into the code 😄 why is it bad to have ddca_enable_verify set to true?

rockowitz, 2022-11-11:
It's not so much that it's bad, it may not be useful depending on how your GUI is structured.  Verification is expensive, entailing a sleep to read the value as well as one when writing it.   It may be redundant if the value is otherwise read immediately after a ddca_set_nontable_vcp_value() call or sequence of calls.  It complicates error handling for ddca_set_nontable_vcp_value(), particularly because it doesn't return the current value.  YMMV

albertvaka, 2023-03-11:
Related to this: I requested abi-laboratory to track libddcutil's releases for ABI changes. In case you are interested it's here: https://abi-laboratory.pro/?view=timeline&l=libddcutil

rockowitz, 2023-03-13:
Thank you for suggesting to abi-laboratory that they track libddccutil.  I look forward to using some of their tools.

What's particularly interesting are the ABI incompatibilities reported for releases 1.2.0 and 1.2.1. The declarations of the affected functions are unchanged, yet the register usage has changed.  I looked at a few of the functions.  What did change was the internal implementation; more specifically the validity checks for the arguments passed.  I hadn't imagined that implementation details could be leaky in a way that affects the ABI.

albertvaka, 2023-03-13:
I was also surprised about that. I don't expect a change to the implementation in a `.c` file to change the ABI... maybe there's something wrong in the analysis @lvc?

This is the diff for the `ddca_free_display_info` function  in [api_displays.c](https://github.com/rockowitz/ddcutil/blob/a33f34155c7efe87a3bcabaa8eef832b08b59b4d/src/libmain/api_displays.c#L813) between `v1.2.0` and `v1.2.1`:

[image]

And this is the ABI change from [the report](https://abi-laboratory.pro/index.php?view=compat_report&l=libddcutil&v1=1.2.0&v2=1.2.1&obj=658ca&kind=abi):

> The parameter **info_rec** became passed in **rbx** register instead of **rdi**.

aleixpol, 2023-03-13:
How can the ABI change if the function declaration is not changing?

aleixpol, 2023-03-31:
> Related to this: I asked abi-laboratory to track libddcutil's releases for ABI changes. You can find it here: https://abi-laboratory.pro/?view=timeline&l=libddcutil

It was Albert who mentioned it earlied.