rockowitz/ddcutil #288: ddcutil sometimes work, sometimes detection error

Gita-Govinda, 2022-11-20:
Hello. I've installed ddcutil in my linux mint desktop. 

When I do _ddcutil detect_ it sometimes detects the monitor and works, sometimes it partially detects and doesn't work. 
On the high level I'm unable to trace a pattern for this random behavior.

Here I've attached _environment_ and _interrogate_ results on both working and non working cases.
[ddcutil-not-working.txt](https://github.com/rockowitz/ddcutil/files/10050525/ddcutil-not-working.txt)
[ddcutil-working.txt](https://github.com/rockowitz/ddcutil/files/10050526/ddcutil-working.txt)

Will appreciate any help. Regards

Gita-Govinda, 2022-11-22:
Thanks rockowitz for the reply,
Seems I'm out of luck with -sleep-multiplier 2.0 and --maxtries "15,15,15" 
Here are the stats and ddc report though. 
[stats.txt](https://github.com/rockowitz/ddcutil/files/10068857/stats.txt)

You're correct, the PC is old, however the monitor is new. Currently it is using a VGA connection.

rockowitz, 2022-11-22:
There's no ***--ddc*** output in your stats.txt file. 

If you have one available, try testing with a different monitor.  There are monitors with marginal DDC implementations.

JFingerle, 2023-05-18:
@rockowitz The "--ddc" output is a the bottom of the stats.txt file

rockowitz, 2023-05-20:
@JFingerle 

I apologize for having missed the ***--ddc*** output at the bottom of the file.  What it shows is that response packets are corrupted.  Several of them have a duplicate initial byte (x6e). Release 1.3.0 added code to test and adjust for the (not uncommon) case of a  duplicate initial byte.