rockowitz/ddcutil #289: patch: crash on libddcutil init with Philips 328E1

giftick, 2022-11-22:
Hello,

I control brightness, red, green and blue of a Philips 328E1 on a linux x86_64 desktop with an X11 GUI application that I link dynamically to libddcutil.

All works fine with ddcutil-0.9.9, but with
- ddcutil 1.2.2 or
- ddcutil 1.3.0 or
- ddcutil git commit f2bc353 (the latest commit as I write this)

my app crashes before it's main() is called (obviously during initialization of libddcutil) with the following error:

```
string_util.c:590: ntsa_length: Assertion 'string_array' failed.
```

The following patch, applied to ddcutil git commit f2bc353, fixes the problem for me:

```
diff --git a/src/libmain/api_base.c b/src/libmain/api_base.c
index 71757cf3..02f80b45 100644
--- a/src/libmain/api_base.c
+++ b/src/libmain/api_base.c
@@ -211,7 +211,6 @@ Parsed_Cmd * get_parsed_libmain_config() {
    //    goto bye;
 
 
-   assert(new_argc >= 1);
    DBGF(debug, "Calling parse_command()");
    parsed_cmd = parse_command(new_argc, new_argv, MODE_LIBDDCUTIL);
    if (!parsed_cmd) {
diff --git a/src/util/string_util.c b/src/util/string_util.c
index ecc6d341..1bc2df20 100644
--- a/src/util/string_util.c
+++ b/src/util/string_util.c
@@ -587,7 +587,7 @@ void ntsa_free(Null_Terminated_String_Array string_array, bool free_strings) {
  *  Equivalent to g_strv_length()
  */
 int ntsa_length(Null_Terminated_String_Array string_array) {
-   assert(string_array);
+   if (string_array == NULL) return 0;
    int ndx = 0;
    while (string_array[ndx] != NULL) {
       ndx++;
@@ -653,7 +653,7 @@ Null_Terminated_String_Array ntsa_join(
  */
 Null_Terminated_String_Array ntsa_copy(Null_Terminated_String_Array a1, bool dup)
 {
-   assert(a1);
+   if (a1 == NULL) return NULL;
    int ct = ntsa_length(a1);
    Null_Terminated_String_Array result = calloc((ct+1), sizeof(char *));
    char ** to = result;
```

Best regards and thank you for ddcutil,
Giftick

giftick, 2022-11-23:
Yes, I had this ~/.config/ddcutil/ddcutilrc file when the error occurred:

```
[global]
options: --sleep-less --sleep-multiplier .025 --timeout-i2c-io

[ddcutil]

[libddcutil]
--noverify
```

When I remove this config file, ddcutil git commit f2bc353 works fine without my patch.
I don't need the config file anyway. My app calls
```
ddca_enable_report_ddc_errors(false);
ddca_enable_verify(false);
ddca_enable_sleep_suppression(true); // I know this is a noop nowadays
```
on startup, and then does this in an async thread that processes queued change requests in a loop:
```
__useconds_t sleep_usec = 45000; // Philips 328E1: 45000 always works, 44000 occasionally drops a packet
for (;;) {
    ...
    DDCA_Status rc = ddca_set_non_table_vcp_value(ddca_dh,code,v >> 8,v & 0xff);
    if (rc) error(...);
    usleep(sleep_usec);
}
```

Sorry for not having thought about the config file myself.

rockowitz, 2022-11-24:
I have pushed a fix for the crash you reported to branch 1.4.0-dev.  The underlying problem was that function apply_config_file() in file ddcutil_config_file.c was not setting the proper return values for the case where the config file  is invalid.  The assert() failure in function ntsa_length() is a precondition failure.

The configuration file error that triggered the problem is the [libddcutil] segment, which should read "options: --noverify", 
not "--noverify". 

Thank you for reporting the problem. Getting all the bugs out of error handling code is difficult, since execution paths through error handlers are infrequent.

I have also modified the libddcutil initialization code to write the error message to the system log as well as syserr, since messages to syserr can be lost depending on how the program using the shared library is invoked. As the libddcutil initialization function __ddca_init() is a constructor it doesn't set the thread error detail that can be checked by the caller as with other API functions.  Perhaps there needs to be an additional API function ddca_get_initialization_msgs() to get this information.   Thoughts?

I'm interested to see how you're using libddcutil, but I'd rather not clutter this bug thread.  If your code is on github I'll take a look.  Or you can email me at rockowitz@minsoft.com.

Regards, 
Sanford

rockowitz, 2023-01-18:
Fixed in release 1.4.1

closed 2023-01-18 by rockowitz