rockowitz/ddcutil #290: USB-C to HDMI adapter with DDC communication failures

prodrigestivill, 2022-11-25:
Hello,

I'm using an USB-C to HDMI adapter and there are some errors with the DDC communication:
```
Invalid display
[...]
DDC communication failed
```

If I switch it to HDMI to HDMI cable it works correctly, so it is no issue with the Monitor, I assume it might be with the Intel DRM driver or the cable or the timings in the ddcutil.

In any case I wasn't able to find a similar enough issue and non of the other workarounds found in the issues did work.

- I'm already using kernel >5.10 ([issue drm-intel #37](https://gitlab.freedesktop.org/drm/intel/-/issues/37)).
- I already tried `--force-slave-address` with the latest git from yesterday with no success.

So I decided to create this new issue and I attach the output form [ddcutil interrogate](https://github.com/rockowitz/ddcutil/files/10093501/ddcutil-interrogate-git.txt) from yesterday's git version.

rockowitz, 2022-11-25:
First, thank you for creating a new issue.

My best guess is that it's the conversion of the I2C signal in the USB-C to HDMI adapter.   The video related signals in USB-C are the same as DisplayPort, using the DisplayPort Aux channel. (For details on I2C on DisplayPort and conversion to HDMI, see section [DDC Over DisplayPort](https://www.ddcutil.com/displayport/l).

A couple things you might try to confirm this: 
- Try a different UBC-C->HDMI adapter
- If you have a USB-C->DisplayPort adapter, try using that to connect to the DisplayPort input on the monitor (which I believe your monitor has) 
- Use a USB-C->DisplayPort adapter, along with an "active" DisplayPort->HDMI cable or adapter.  The signal conversion would then occur in the DisplayPort->HDMI cable.

closed 2022-11-28 by rockowitz

prodrigestivill, 2022-11-29:
Hello @rockowitz, awesome explanation in the documentation, not sure why I didn't manage to find it before.
Thanks a lot!

Note: if you can edit the last comment to remove the typo in the link [URL](https://www.ddcutil.com/displayport/), it might be helpful for the next user having the same issue.

rockowitz, 2022-11-29:
I'm glad the documentation helped. Thank you for pointing out the typo.