rockowitz/ddcutil #291: ddcutil probe fails on Ubuntu 22.04 & 22.10 with AMD Ryzen 6900HX CPU

parke, 2022-12-16:
`ddcutil probe` is failing when run on Ubuntu 22.04 (and 22.10) on an AMD Ryzen 6900HX CPU.  (Kernels 5.15 and 5.19.)

As the 6900HX is a new CPU, my initial thought was that I needed a newer kernel (and/or newer AMD drivers).

However:

`ddcutil probe` succeeds when run on EndeavorOS (i.e. Arch Linux) on the 6900HX with both a 5.15 and 6.0 kernel.

`ddcutil probe` also succeeds on Ubuntu 22.04 with an older Ryzen 5600G CPU (on a different motherboard).

The particular way `ddcutil probe` fails on the 6900HX is also interesting.  It just prints `Killed` and exits.

Also, if I run `ddcutil interrogate` on Ubuntu 22.10 on the Ryzen 6900HX CPU, `ddcutil` blocks forever and cannot be killed.  And eventually (after 5 to 20 minutes?) the screen and keyboard freeze, and the system stops responding to network pings.

Do you have any advice for how to diagnose this problem further?

I should add that I am a newbie user of `ddcutil`.  I've used `ddccontrol` for years.  However, I am submitting this issue here as Arch Linux provides a compiled `ddcutil` package, but does not provide a compiled `ddccontrol` package.

Here is a summary of what I have observed.

**Case 1  -  `ddcutil probe` fails.
                 `ddcutil interrogate` blocks indefinetily.**
               
CPU:      AMD Ryzen 6900HX with Radeon Graphics
Distro:   Ubuntu 22.10 (or 22.04)
Kernel:   linux-image-generic  5.19.0.26.23
ddcutil:  1.3.0-1

```
$ sudo ddcutil probe
[sudo] password for user:
Killed

$ sudo ddcutil interrogate
Setting output level very-verbose...
Setting maximum retries...
Forcing --stats...
Forcing -disable-capabilities-cache
This command will take a while to run...

[ ddcutil runs forever? ]
[ ddcutil cannot be interrputed with ctrl-C. ]
[ ddcutil cannot be killed via kill. ]
[ After 5 to 20 minutes, the system freezes. ]

```

**Case 2  -  `ddcutil probe` runs fine.**

CPU:      AMD Ryzen 6900HX with Radeon Graphics
Distro:   EndeavorOS
Kernel:   linux-lts  5.15.83-1
Kernel:   linux      6.0.12.arch1-1
ddcutil:  1.3.0-1

**Case 3  -  `ddcutil probe` runs fine.**

CPU:      AMD Ryzen 5 5600G with Radeon Graphics
Distro:   Ubuntu 22.04
Kernel:   linux-image-generic  5.15.0.53.53
ddcutil:  1.2.2-1

parke, 2022-12-16:
Today I tested with a Debian 11 Live ISO and a Pop_OS 22.04 installation.

On Debian 11:  The kernel was 5.10.  ddcutil was a version below 1.0, maybe 0.6.  ddcutil printed an error message saying that it failed to detect the display.

On Pop_OS 22.04:  The kernel is version 6.0.6-76060006.  ddcutil is version 1.2.2.  Happily, ddcutil seems to work on this installation of Pop_OS.  Also happily, ddccontrol works!  

I am going to close this issue, as I am willing to install and use Pop_OS.

closed 2022-12-16 by parke

rockowitz, 2022-12-18:
I've been giving the problem you encountered some thought.  In all likelihood it's a system call that doesn't return.  I imagine there's a bug in the amdgpu driver's handling of the Radeon 680M that's in the 6900HX.  I've seen this in the past, for example issue #194.  In that case, a fix went into the kernel (5.17 IIRC), but I was also able to put a workaround into ddcutil 1.1.0.  

If you can, please execute the a ddcutil command that hangs using both strace and the ddcutil tracing facility  The output will be voluminous, but the last 100 lines or so should be able to pinpoint the system call that's failing.  If you want to direct the output to a file, note that strace writes to stderr, not stdout.  For example
`$ strace ddcutil interrogate > myfile.txt 2>&1`

FYI, **interrogate** is basically a convenience command that includes the output  of **detect --verbose**, **environment --verbose**, and for each monitor **probe**.

parke, 2022-12-18:
Update:  So what happens is:

The first time I run `ddcutil interrogate`, it is killed.
The second time I run `ddcutil interrogate`, it blocks forever.
(The tail of both strace logs is below.)

Also, when I do a system restart after the second run of `ddcutil interrogate`, a call trace is printed to the screen.

The call trace includes:

```
[snip]
[ 345.778404] ---[ end trace 000000000000000 ]---
[ 345.798110] RIP: 0010:amdgup_dm_i2c_xfer+0xe2/0x140 [amdgpu]
[ 345.798536] Code b7 54 [snip]
[snip]
```

If you want, I can email you the full strace logs.  gzipped, they are only 19KB.  I can also email you photos of the shutdown call trace.  Only part of the call trace is visible in the photos.

These tests were run on an Ubuntu 22.10 Live ISO.  I have installed Fedora 37 on the system, and am in the process of configuring it for regular use.  Once the system configured and deployed, it will be inconvenient for me to reboot it to perform additional tests.  Deployment could happen within the next 24 hours.  I only have one system with the Ryzen 6900HX CPU.

First run:

```
[snip]
readlink("/sys", 0x7ffce8b93f20, 1023)  = -1 EINVAL (Invalid argument)
readlink("/sys/bus", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/i2c", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/i2c/devices", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/i2c/devices/i2c-5", "../../../devices/pci0000:00/0000"..., 1023) = 59
readlink("/sys/devices", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0/i2c-5", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0/i2c-5/device", "../../0000:34:00.0", 1023) = 18
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0", 0x7ffce8b93f20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys", 0x7ffce8b93f60, 1023)  = -1 EINVAL (Invalid argument)
readlink("/sys/devices", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0/driver", "../../../../bus/pci/drivers/amdg"..., 1023) = 34
readlink("/sys/bus", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/pci", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/pci/drivers", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/pci/drivers/amdgpu", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/pci/drivers/amdgpu/module", "../../../../module/amdgpu", 1023) = 25
readlink("/sys/module", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/module/amdgpu", 0x7ffce8b93f60, 1023) = -1 EINVAL (Invalid argument)
openat(AT_FDCWD, "/dev/i2c-5", O_RDWR)  = 3
ioctl(3, _IOC(_IOC_NONE, 0x7, 0x5, 0), 0x7ffce8b974e0) = 0
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
ioctl(3, _IOC(_IOC_NONE, 0x7, 0x7, 0), 0x7ffce8b97370) = -1 EIO (Input/output error)
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
ioctl(3, _IOC(_IOC_NONE, 0x7, 0x7, 0), 0x7ffce8b97370) = -1 EIO (Input/output error)
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
ioctl(3, _IOC(_IOC_NONE, 0x7, 0x7, 0), 0x7ffce8b97370) = -1 EIO (Input/output error)
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-5", 4096) = 10
[... 110 lines cut ...]
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0/i2c-6", 0x7fff9ce00a20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0/i2c-6/device", "../../0000:34:00.0", 1023) = 18
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0", 0x7fff9ce00a20, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys", 0x7fff9ce00a60, 1023)  = -1 EINVAL (Invalid argument)
readlink("/sys/devices", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:34:00.0/driver", "../../../../bus/pci/drivers/amdg"..., 1023) = 34
readlink("/sys/bus", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
[... cut ...]
readlink("/sys/module", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
readlink("/sys/module/amdgpu", 0x7fff9ce00a60, 1023) = -1 EINVAL (Invalid argument)
openat(AT_FDCWD, "/dev/i2c-6", O_RDWR)  = 3
ioctl(3, _IOC(_IOC_NONE, 0x7, 0x5, 0), 0x7fff9ce03fe0) = 0
readlink("/proc/self/fd/3", "/dev/i2c-6", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-6", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-6", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-6", 4096) = 10
readlink("/proc/self/fd/3", "/dev/i2c-6", 4096) = 10
ioctl(3, _IOC(_IOC_NONE, 0x7, 0x7, 0)
```

rockowitz, 2022-12-18:
First, let me try to refine the tests.  Please send the complete zipped output, and whatever relevant screenshots you have.
~~~
$ strace ddcutil interrogate --trace all > file1 2>&1
$ strace ddcutil detect --trace all > file2 2>&1
~~~
Googling the location (amdgup_dm_i2c_xfer) of the failure, I came on the following [patch](https://www.spinics.net/lists/amd-gfx/msg80168.html) related to an eDP device (i.e. a laptop display).  I have a hunch that the hang occurs when ddcutil examines (i.e. sends a DDC command) on the I2C bus for the eDP device.  If you use the **--bus** option to specify the i2c bus for your external monitor,  **ddcutil** will not examine any other bus. If for example the I2C bus for your monitor is /dev/i2c-5, the following command would not touch the I2C bus for the laptop's display: 
~~~
$ ddcutil getvcp 10 --bus 5
~~~

Also, can you describe your hardware configuration?   Laptop model, external display, docking station or DisplayPort MST? 

Thanks.

rockowitz, 2022-12-20:
Parke,  please send on the additional logs.  Given your hardware configuration, the my hunch re eDP was off base.  I can see that the hang occurs in amdgpu.  I'm hoping the additional ddcutil trace information may help me create a simple  program that triggers the bug.  

What /dev/i2c buses are used for the displays?

parke, 2022-12-22:
The system has two HDMI ports.  However, I only attach one display at a time.  I am assuming that the display is attached to `/dev/i2c-6`, as that is the device that is causing the `SIGKILL` and blocked process.  (The other HDMI port may use a different `i2c` device.)

I sent you the additional logs you requested via email.

rockowitz, 2022-12-23:
Based on the logs you sent me, a LG monitor is found at /dev/i2c-4.  However, the failure occurs when trying to read an EDID from /dev/i2c-6.  So the question is what device is on /dev/i2c-6.

Please execute the following two commands and submit the logs: 

~~~
$ grep ^ /sys/bus/i2c/devices/*/name
~~~

~~~
$ ddcutil getvcp 10 --bus 4 --f2
~~~

As explanation for what is going on here, **--f2** is an option for executing some undocumented behaviour, typically for testing.  It currently probes /sys for I2C device information.  (This information is normally part of the output from  command **environment --verbose**). **--bus 4** skips display detection and the command should avoid touching /dev/i2c-6

rockowitz, 2022-12-24:
@parke  Your most recent email reports that your "system is no longer available for reports into Ubuntu 22.04 or 22.10 where the problems occurred". 

To summarize, my best guess is that the problem is similar to that reported in issue #194.   It may be that the other distributions you have tested are running more recent kernel versions in which a bug is fixed.

For those encountering similar problems, I have two suggestions, as described above: 

1) Try using the ***--bus*** option on the **getvcp** and **setvcp** commands to avoid bus scanning.

2) The following command shows the names of i2c devices.  See if the hang occurs on a i2c device with an unusual name: 
~~~
$ grep ^ /sys/bus/i2c/devices/*/name
~~~

parke, 2022-12-24:
@rockowitz Actually my email said that:  "The system is no longer available for **reboots** into Ubuntu 22.04 or 22.10 where the problems occurred."

As you mention, the system is now running a 6.0 kernel and the problem is not present.

Thank you for your assistance diagnosing this problem.