rockowitz/ddcutil #292: ddcutil detect (version 1.2.2) ignores --bus

parke, 2022-12-31:
As of ddcutil version 1.2.2:

```
$ ddcutil --bus 4 --terse detect
Open failed for /dev/i2c-3: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-5: errno=EACCES(13): Permission denied
Display 1
   I2C bus:  /dev/i2c-4
   Monitor:             GSM::
```

My use-case is that I `chmod 666 /dev/i2c-4` so that non-root users can control the monitor.  At present, non-root users can detect the monitor (as shown above), but I'd prefer to avoid generating unnecessary `EACCES` warnings.  I detect the monitor model in order to apply brightness and contrast settings on a per-model basis.  (All as non-root.)

I have not checked to see if `ddcutil detect` has already started respecting `--bus` in versions above 1.2.2.

rockowitz, 2023-01-01:
**detect** ignores ***--bus*** because the device selection options apply only to commands that operate on a single display, not those that operate on multiple displays or no displays.  In general, most options apply only to some subset of commands, and are ignored if inapplicable.  However, I have added a check in the parser to explicitly reject the display selection options on commands to which they do not apply. 

The EACCES errors are reported because it makes clear that there's a permissions problem.  Users need a clue as to why a **ddcutil** command doesn't work.

Why do you want to perform monitor detection if you already know the I2C bus to use?  What is the use case.

Until now, the recommended way give-non root users access to the I2C devices is as outlined in the I[2C Device Permissions](http://www.ddcutil.com/i2c_permissions/) page of [www.ddcutil.com](http://www.ddcutil.com). In short, all /dev/i2c devices are assigned to group **i2c**, and have read/write access to those devices.  This happens automatically in **ddcutil** packages for Debian and derived distributions such as Ubuntu, because package **ddcutil** depends on package **i2c-tools**.  All that remains is for **ddcutil** users to be added to group **i2c**.  

Branch 1.4.0-dev simplifies permissions by adding a udev rule that uses the **uaccess** tag to give the logged on user read/write access to all /dev/i2c devices associated with video adapters.   See issue #275. Installing package **ddcutil** installs file /usr/lib/udev/rules.d/60-ddcutil.rules.  For testing without package installation, copy file data/usr/lib/udev/rules.d/60-ddcutil.rules to directory /etc/udev/rules.d.

parke, 2023-01-01:
> **detect** ignores _**--bus**_ because the device selection options apply only to commands that operate on a single display, not those that operate on multiple displays or no displays. In general, [...]

Understood.  What I was suggesting was a mechanism (via `--bus`) for allowing `detect` to operate on a single i2c device.
 
> The EACCES errors are reported because it makes clear that there's a permissions problem. Users need a clue as to why a **ddcutil** command doesn't work.

Yes, I understood this.  And it the expected and correct behavior, in my opinion.

> Why do you want to perform monitor detection if you already know the I2C bus to use? What is the use case?

As I said originally:  "I detect the monitor model in order to apply brightness and contrast settings on a per-model basis."

I will elaborate:

My script can set the attached monitor to what I call `high`, `medium`, or `low` brightness and contrast settings.  The particular brightness and contrast numbers I use are different for each monitor.

At present, my scripts infer the monitor model based on the hostname.  So, for example, my HP 2009m monitor is connected to host `alpha`, whereas my LG monitor is connected to host `beta`.  Also, the LG has a gamma setting, whereas the HP does not.  So, on host `beta` my script sets the (inferred) LG monitor to a brightness, contrast, gamma of: `70 70 1`, `50 70 0`, or `40 40 0`.  Whereas on host `alpha`, the (inferred) HP monitor gets set to one of: `50 90`, `50 35`, or `30 30`.

However, there are cases where the hostname-model inference is incorrect.  Sometimes I connect `alpha`'s monitor to `beta` (or vice versa).  And sometimes I create a new host whose name is neither `alpha` nor `beta`.  In these cases, I have to manually edit my script every time I switch monitors or create a system with a new hostname.

> Until now, the recommended way give-non root users access to the I2C devices is as outlined in the I[2C Device Permissions](http://www.ddcutil.com/i2c_permissions/) page of [...]

Thank you for suggesting using the `i2c` group.  This is probably a satisfactory solution for me.

However, I still do not see the downside of allowing `detect` to be focused on a single device via the `--bus` option.

Up until a week ago, I was using `ddccontrol` to adjust my displays' contrast and brightness.  Probing with `ddccontrol` was a slow and time consuming operation (5 to 20 seconds?).  Furthermore, as it happens, my HP 2009m monitor sometimes fails to respond to DDC instructions (including detecting and probing).  (Such failures can happen with both `ddccontrol` and `ddcutil`.)  So probing was not 100% reliable.

`ddcutil detect` seems to be much faster than `ddccontrol -p`.  I'm not sure if `ddcutil detect` is 100% reliable at detecting my HP 2009m.  I know I have observed cases where attempting to change the brightness or contrast of the HP 2009m via `ddcutil` results in `ddcutil` reporting a failure.  In these cases, my script waits a fraction of a second and then tries again.

So my script is designed around the limitations of `ddccontrol` and my HP 2009m.  My script only needs to detect the monitor once per reboot, and then marks the monitor's device by `chmod`ing the device to `666`.  Subsequent invocations of the script "detect" the monitor by looking for the device with permissions of `666`.

> Branch 1.4.0-dev simplifies permissions by adding a udev rule that uses the **uaccess** tag to give the logged on user read/write access to all /dev/i2c devices associated with video adapters. See [...]

Nifty.  I look forward to that feature being present in future releases of Debian and Ubuntu.

Please feel free to close this issue if you prefer not to extend `ddcutil detect` to work with `--bus`.

rockowitz, 2023-01-02:
Modifying the **detect** command to check just a single bus would be non-trivial to implement, both because of the way the code is structured and because of a need for consistency in behaviour with other ways of specifying a single monitor (e.g. ***--model***) and ideally monitors that use USB instead of DDC/CI to communicate with the monitor's Virtual Control Panel.  I'm open to the possibility, but it needs to solve a problem that cannot otherwise be addressed.  

You could, for example,  Just parse the output of **detect** to locate monitors and to associate their models with an I2C bus.  Option ***--brief*** reports monitor information in a more machine readable format. 

Re intermittent failures with your HP 2009m, keep in mind that DDC/CI is an inherently unreliable protocol.  **ddcutil** performs retries, but that is not always successful. Options ***--stats*** and ***-ddc*** are useful to see what kind of failures are occurring. I would suggest trying using option ***--sleep-multiplier*** to increase the time that ddcutil that gives the monitor to formulate and return a reply packet.  This often allows marginal monitors to communicate reliably. ***--sleep-multiplier 2.0*** is a good initial value to try.

parke, 2023-01-02:
Thank you for the response and for the suggestions.

closed 2023-01-02 by parke