rockowitz/ddcutil #293: Ddcutil broken on 2 monitors after 1.4.0 release

Copysiper, 2023-01-14:
My setup:
3 monitors, before the update i could control monitor brightness on any of them.
Now it works only on the primary.

I'm running Wayland+Kde+Nvidia Propietary drivers (which is known to be a bad combo, but worked on a previous version, right?)

Here's the output of ddcutil detect -v on 1.3.0:
```
❯ ddcutil detect --show-table -v
Unable to open directory /sys/bus/i2c/devices/i2c--1: No such file or directory
Device /dev/i2c-255 does not exist. Error = ENOENT(2): No such file or directory
/sys/bus/i2c buses without /dev/i2c-N devices: /sys/bus/i2c/devices/i2c-255
Driver i2c_dev must be loaded or builtin
See https://www.ddcutil.com/kernel_module
Display 1
   I2C bus:  /dev/i2c-3
      DRM connector:                      card0-HDMI-A-1
      Driver:                             nvidia
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-3/name     NVIDIA i2c adapter 5 at 1:00.0
      PCI device path:                    /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/i2c-3
   EDID synopsis:
      Mfg id:               GSM - Goldstar Company Ltd (LG)
      Model:                LG IPS FULLHD
      Product code:         23224  (0x5ab8)
      Serial number:        
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2014,  Week: 1
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.645, 0.335
      Green x,y:        0.305, 0.630
      Blue  x,y:        0.148, 0.050
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 1e 6d b8 5a 01 01 01 01   .........m.Z....
      +0010   01 18 01 03 80 30 1b 78 ea 31 35 a5 55 4e a1 26   .....0.x.15.UN.&
      +0020   0c 50 54 a5 4b 00 71 4f 81 80 95 00 b3 00 a9 c0   .PT.K.qO........
      +0030   81 00 81 c0 90 40 02 3a 80 18 71 38 2d 40 58 2c   .....@.:..q8-@X,
      +0040   45 00 e0 0e 11 00 00 1e 00 00 00 fd 00 38 4b 1e   E............8K.
      +0050   53 0f 00 0a 20 20 20 20 20 20 00 00 00 fc 00 4c   S...      .....L
      +0060   47 20 49 50 53 20 46 55 4c 4c 48 44 00 00 00 ff   G IPS FULLHD....
      +0070   00 0a 20 20 20 20 20 20 20 20 20 20 20 20 01 64   ..            .d
   VCP version:         2.1
   Controller mfg:      Novatek
   Firmware version:    3.2
   Monitor returns DDC Null Response for unsupported features: false

Display 2
   I2C bus:  /dev/i2c-4
      DRM connector:                      card0-DP-1
      Driver:                             nvidia
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-4/name     NVIDIA i2c adapter 6 at 1:00.0
      PCI device path:                    /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/i2c-4
   EDID synopsis:
[... 200 lines cut ...]
      +0040   65 0c 58 54 21 00 00 1a 28 68 00 a0 f0 70 3e 80   e.XT!...(h...p>.
      +0050   08 90 65 0c 58 54 21 00 00 1a 00 00 00 fd 00 28   ..e.XT!........(
      +0060   3d 87 87 38 01 0a 20 20 20 20 20 20 00 00 00 fc   =..8..      ....
      +0070   00 4c 47 20 55 6c 74 72 61 20 48 44 0a 20 01 21   .LG Ultra HD. .!
   VCP version:         2.1
Extended delay as recovery from DDC Null Response...
   Controller mfg:      Mstar
Extended delay as recovery from DDC Null Response...
   Firmware version:    3.23
   Monitor returns DDC Null Response for unsupported features: false
```

When trying to change brightness on previously working displays it either says

`Display not found `

or

```
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
Setting value failed for feature 10, rc=DDCRC_RETRIES(-3007): maximum retries exceeded
    Try errors: DDCRC_RETRIES
```

I hope that info helps, I am ready to provide more if you need something else, thanks in advance!

rockowitz, 2023-01-14:
First, please run **ddcutil interrogate** and send the output.  That preempts lots of back and forth on basic questions.

The most obvious suspect is an error in the new code to workaround a bug in the way the nvidia driver is coded.
Some options to experiment with:  ***--use-file-io***, ***--use-ioctl-io***.    Any effect?

The fact that communication with the LG monitor succeeds, but with error recovery, suggests that the problem is happening at the I2C level.  Let's look at the communication layer in detail.  Execute **ddcutil detect --verbose --ddc --trcfile i2c_strategy_dispatcher.c  --trcfile i2c_execute.c**.  Does using option ***--sleep-multiplier*** with a value greater than 1, e.g. ***--sleep-multiplier 2***, have any effect?

Are you using the [Special Nvidia Driver Settings](https://www.ddcutil.com/nvidia/)?

@fionnb has reported what appears to be a similar regression as an additional comment on issue #273.  

Please submit long output as files of some sort, e.g. as attachments, instead of inline.  

Thank you for your help in diagnosing this situation.  I'd like to quickly resolve the problem.  The cutoffs for including **ddcutil** release 1.4 in the upcoming Debian release (bookworm) and in Ubuntu 23.04 are less than a month away.

fionnb, 2023-01-14:
Quick update from my side: Problem is with ab LG monitor too. But X11 here and vanilla intel-mesa. So probably not an issue with the GFX drivers. Will gather debug data tomorrow and send another update.

Copysiper, 2023-01-15:
OK, so here are the outputs:
 Quickly looking at them, it looks like appending --sleep-multiplier 2 for the last command actually seems to improve the situation, resulting in "Invalid display" header to disappear for them become "Display 1", "Display 2", "Display 3",
after that i tried ddcutil setvcp 10 for all 3 monitors with --sleep-multiplier 2 and all of them worked. However, is there an option to change that setting by default?

[interrogate.log](https://github.com/rockowitz/ddcutil/files/10419931/interrogate.log)
[detect.log](https://github.com/rockowitz/ddcutil/files/10419930/detect.log)
[detect_with_sleep_2.log](https://github.com/rockowitz/ddcutil/files/10419929/detect_with_sleep_2.log)

rockowitz, 2023-01-15:
What I'm seeing is corrupt DDC data packets, which is resolved by longer waits between I2C read and write operations.  Why this effect only appears with release 1.4.0  is unclear.  I'll likely get back to you with additional tests.

A default value for ***--sleep-multiplier*** can be set using a [Configuration File](https://www.ddcutil.com/config_file/#configuration-file).  

I suggest experimenting to see if a sleep multiplier value less than 2, e.g. 1.5, works for you. Options ***--ddc*** and ***--stats*** can be helpful in that they'll show the extent that errors are recovered using retry.

Copysiper, 2023-01-15:
Well, after playing with parameters I got that it's pretty unstable if sleep value is <=0.3, so I chose 0.5, which is quite stable for me(no errors after many retries). Changing config worked.

However, sadly, changing config resolved issue with cli only, but not with powerdevil-ddcutil, which uses ddcutil to help with automating lowering brightness when AFK on KDE. I think powerdevil ignores/overrides config parameters, so I'll stay on 1.3.0 for now

rockowitz, 2023-01-15:
@copysiper I'm a bit confused here.  Your original post indicated that ddcutil failed on release 1.4.0, and it turned out that increasing sleep time using ***--sleep-multiplier 2*** resolved the problem.  There's no indication that you by default were using a sleep multiplier value other than 1.    If a value of 2 succeeds but 1 fails, it makes no sense for .5 to work. Can you clarify?

Re powerdevil,  it uses shared library **libddcutil**, not command line program **ddcutil**.  To pass options to the shared library, use the [libddcutil] section of configuration file ddcutilrc.

Copysiper, 2023-01-16:
Actually, for me it behaves like if --sleep-multiplier 0 is set by default if nothing specified in arguments or config, so when i add --sleep-multiplier 1 in config or arguments it starts working too

Copysiper, 2023-01-16:
Ok, so i cleared the config and updated my package to 1.4.1 (https://github.com/rockowitz/ddcutil/commit/0dec3a4e9714e2f8ee4b93ed629c00ea10979d82),
ddcutil cli definitely works, sleep multiplier now behaves like if it is 1 by default, no errors appear when changing brightness on any display.
powerdevil is able to manage brightness correctly too, here is the log of the correct version of the lib being initialized:
```
libddcutil[3749]: Initializing.  ddcutil version 1.4.1
libddcutil[3749]: Library initialization complete.
```
It looks like the issue is resolved now, thank you for such fast support!

hesdago, 2023-01-16:
Hello, and thanks for supporting on this.
I'm seeing similar issues after upgrading from 1.3 to 1.4, and the mentioned parameter for sleep-multiplier does help:
`--sleep-multiplier 2`

```
➜ dago@dago-pc-amd64  ~  sudo ddcutil detect
Display 1
   I2C bus:  /dev/i2c-1
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL U2715H
      Product code:         53351  (0xd067)
      Serial number:        GH85D7CM0T8S
      Binary serial number: 810825811 (0x30543853)
      Manufacture year:     2017,  Week: 51
   VCP version:         Detection failed

Display 2
   I2C bus:  /dev/i2c-3
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL U2715H
      Product code:         53349  (0xd065)
      Serial number:        GH85D81801PL
      Binary serial number: 808538188 (0x3031504c)
      Manufacture year:     2018,  Week: 2
   VCP version:         2.1

➜ dago@dago-pc-amd64  ~  sudo ddcutil detect --sleep-multiplier 2
Display 1
   I2C bus:  /dev/i2c-1
   DRM connector:           card0-HDMI-A-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL U2715H
      Product code:         53351  (0xd067)
      Serial number:        GH85D7CM0T8S
      Binary serial number: 810825811 (0x30543853)
      Manufacture year:     2017,  Week: 51
   VCP version:         2.1

Display 2
   I2C bus:  /dev/i2c-3
   DRM connector:           card0-DP-1
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL U2715H
      Product code:         53349  (0xd065)
      Serial number:        GH85D81801PL
      Binary serial number: 808538188 (0x3031504c)
      Manufacture year:     2018,  Week: 2
   VCP version:         2.1

```

fionnb, 2023-01-16:
I confirm that the "Verification failed" issue has gone away with 1.4.1-dev

closed 2023-02-02 by rockowitz