rockowitz/ddcutil #296: Change input source on Acer X34 Predator

Gustav0ar, 2023-02-02:
Anyone could help me change input source on this old monitor of mine? It seems to not have feature 60, try a lot options but no luck.

```
Unparsed capabilities string: (prot(monitor)type(lcd)model(Acer X34)cmds(01 02 03 06 07 0C E3 F3)vcp(02 03(01) 04 05 08 10 12 14(03 05 09 0B) 16 18 1A 2E 52 59 5A 5B 5C 5D 5E 72(00 78 FF) 8A AC AE B6 C0 C8 C9 CA DF)mccs_ver(2.2)vcpname(10(Brightness))mswhql(1))
Model: Acer X34
MCCS version: 2.2
Commands:
   Op Code: 01 (VCP Request)
   Op Code: 02 (VCP Response)
   Op Code: 03 (VCP Set)
   Op Code: 06 (Timing Reply )
   Op Code: 07 (Timing Request)
   Op Code: 0C (Save Settings)
   Op Code: E3 (Capabilities Reply)
   Op Code: F3 (Capabilities Request)
VCP Features:
   Feature: 02 (New control value)
   Feature: 03 (Soft controls)
      Values (unparsed): 01
      Values (  parsed):
         01: Button 1 active
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values (unparsed): 03 05 09 0B
      Values (  parsed):
         03: 4000 K
         05: 6500 K
         09: 10000 K
         0b: User 1
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 2E (Gray scale expansion)
   Feature: 52 (Active control)
   Feature: 59 (6 axis saturation: Red)
   Feature: 5A (6 axis saturation: Yellow)
   Feature: 5B (6 axis saturation: Green)
   Feature: 5C (6 axis saturation: Cyan)
   Feature: 5D (6 axis saturation: Blue)
   Feature: 5E (6 axis saturation: Magenta)
   Feature: 72 (Gamma)
      Values (unparsed): 00 78 FF
      Full range of absolute adjustment supported (0xff)
      Absolute tolerance: ideal (=0x00)
      Native gamma: 2.20 (0x78)
   Feature: 8A (Color Saturation)
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B6 (Display technology type)
   Feature: C0 (Display usage time)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: CA (OSD/Button Control)
   Feature: DF (VCP Version)
```

[acerx34interrogate.txt](https://github.com/rockowitz/ddcutil/files/10563923/acerx34interrogate.txt)

rockowitz, 2023-02-02:
If the monitor doesn't support feature x60 (Input Source) I'm afraid you're out of luck. Occasionally a  monitor supports a feature without out declaring it in the capabilities string but, consistent with the capabilities string, the last (**probe**) segment of the **interrogate** output fails when attempting to read the value of x60.

I can also see from the **probe** output that the monitor never returns a valid reply indicating that a feature is unsupported, which is why you see the all the "Maximum retries exceeded" messages.  I'd say the monitor's DDC implementation is poorly done.

Gustav0ar, 2023-02-02:
> If the monitor doesn't support feature x60 (Input Source) I'm afraid you're out of luck. Occasionally a monitor supports a feature without out declaring it in the capabilities string but, consistent [...]

I was afraid you were going to say that, thanks for clarification though.

closed 2023-02-02 by Gustav0ar