rockowitz/ddcutil #299: About release 1.4.2

chenrui333, 2023-02-23:
👋 while I can see [the release artifact for 1.4.2](https://www.ddcutil.com/tarballs/ddcutil-1.4.2.tar.gz) on [this page](https://www.ddcutil.com/releases/), but I can see the git tag or release notes for 1.4.2, raise this issue to confirm if that is intended. Thanks!

relates to https://github.com/Homebrew/homebrew-core/pull/123803

rockowitz, 2023-02-23:
While some preliminary artifacts for release 1.4.2 are visible, I do not expect to finalize release 1.4.2 until it is accepted into Debian, as it addresses a Debian bug report.  When it is released, there will be announcements and documentation in the usual places.

chenrui333, 2023-02-24:
make sense, thanks for confirming it.

chenrui333, 2023-05-20:
just a quick checking, looks like it still did not get accepted into debian, any PR that I can also follow? Thanks!

rockowitz, 2023-05-21:
1.4.2 didn't make it through the queue before the code freeze for Debian bullseye, and did not justify a freeze exception  The only change from 1.4.1 was the addition of a file to /etc/modules-load.d to ensure that driver i2c-dev is loaded, relieving the user of a potential configuration step.  Development has long since passed 1.4.2 by, and it's not worth the effort that the Debian approval process takes. The next release will be 2.0.0, which as the numbering suggests is a major one. I'm hoping to put out a release candidate late June.

If you're interested, do give the 2.0.0-dev branch a try.  The CHANGELOG.md file is unfinished, but will give you an idea of what's in it.  Given the extent of changes, user testing will be important for this release. Feedback is very much appreciated.

chenrui333, 2023-05-29:
Thanks for the info! Happy to test RC release later. :) Gonna close this issue for now.

closed 2023-05-29 by chenrui333