rockowitz/ddcutil #300: USB-C to HDMI communication failure

Fushko, 2023-03-10:
Hello,

DDC communication fails from my laptop. I suspect the reason is that I'm using an USB-C-to HDMI adapter, since from my desktop, through the DP port, the monitor is detected and I'm able to set VCP values without any issues.

I have read the FAQ concerning the docking station issues. I'm not sure how relevant it is for my case, since my kernel version is 6.2, and I'm using a simple [adapter](https://www.amazon.com/TSUPY-Thunderbolt-Aluminium-Converter-Compatible/dp/B092MKZFRC) rather than a dock.  (Perhaps it may be worth mentioning adapters too in the FAQ).

Below is the output of `sudo ddcutil interrogate --verbose` for both setups if that can be of any help.

[ddcutil-interrogate-laptop.txt](https://github.com/rockowitz/ddcutil/files/10941164/ddcutil-interrogate-laptop.txt)
[ddcutil-interrogate-desktop.txt](https://github.com/rockowitz/ddcutil/files/10941165/ddcutil-interrogate-desktop.txt)

This is most definitely related to #290. I'm posting this because that issue is closed. I apologize for the duplication otherwise.

rockowitz, 2023-03-13:
First, thank you for opening a new issue instead of re-using an old one.  Issues frequently get cluttered with what are actually quite different problems/requests etc. 

Second, I have to say that I don't have direct experience using a Type-C connector, and looking at documentation online suggests there's a lot of variability in capabilities and implementation.  So take what I say here with a grain of salt.

From your output, I see that the video driver is using its DisplayPort interface - that is, it thinks it's communicating with a DisplayPort device.  It is not using DP Dual mode,  So the DisplayPort -> HDMI conversion is happening in the adapter.  Given how inexpensive your adapter is, I would not be surprised to find that it doesn't fully convert the I2C signal between the DP Aux channel and the physical pins used in HDMI.

I see that your monitor has a DP input.  I'd try using a Type-C <-> Displayport cable,  since that entails no protocol conversion.

rockowitz, 2023-03-19:
Re the Type-C <-> HDMI cable for the laptop, it would have to include an active, not passive, adapter that properly handles the I2C signals.  This might do: https://www.amazon.com/Accell-USB-C-HDMI-Adapter-Thunderbolt/dp/B01K9S0VYM?th=1

As for the desktop, the connection issues are much simpler.  First, you could use a DVI <-> HDMI cable.  From our perspective, it's just a straight through connection, no signal/protocol conversion needed.  

Alternatively, given how recent it is, the 1660ti almost certainly supports Display Port Dual Mode, aka DP++.  In that case the card recognizes that it's Displayport connector is actually talking to a HDMI connection on the monitor, and reconfigures itself to treat the DP connector as just a just a physically different kind of HDMI output.  In that case any DP<->HDMI cable will do.  If it doesn't, then an active DP<->HDMI is needed.  

See: [DDC Over DisplayPort](https://www.ddcutil.com/displayport/#ddc-over-displayport).  I also found this [reddit post](https://www.reddit.com/r/UsbCHardware/comments/rql63s/are_all_usb_typectohdmi_cables_active/) informative, though the comment stating that the active/passive distinction is irrelevant is misleading.  I'm using active/passive in the sense described in the DisplayPort Interoperability Guide.  Though it may be the case that some cables/converters marketed as "active" in fact. for our purposes, just passive with some powered electronic circuitry. 

Bear in mind that the proprietary Nvidia driver is problematic in its DDC implementation.  See [Special Nvidia Driver Settings](https://www.ddcutil.com/nvidia/).

Fushko, 2023-03-20:
Thank you for the exhaustive response. The topic of monitor cables is a vast and interesting one it seems.
I think I'll just go for the DP <-> HDMI cable. Might be handy to have around in any case.