rockowitz/ddcutil #301: DDC communication failed on Odroid N2+ with Ubuntu 22.04 and ddcutil 1.4.1

pharris2411, 2023-03-12:
Hello! I've been trying to debug this for a bit now. I'm trying to use this on a Ubuntu 22.04 install on [an Odroid N2+](https://www.hardkernel.com/shop/odroid-n2-with-4gbyte-ram-2/) using their official image, and an Acer K272HUL monitor. 

I've tested with both ddcutil version 1.2.2 which is the latest from apt, and then built 1.4.1 (minus USB support). I've attached interrogation results from both versions. I verified DDC control does work with this monitor by testing with [Lunar](https://lunar.fyi/) on my M1 MacBook running MacOS 13.0. I'm guessing there is something odd with the Odroid GPU drivers.

Any advice would be much appreciated, thank you!

[ddcutil.1.2.2.txt](https://github.com/rockowitz/ddcutil/files/10951076/ddcutil.1.2.2.txt)
[ddcutil.1.4.1.txt](https://github.com/rockowitz/ddcutil/files/10951077/ddcutil.1.4.1.txt)

rockowitz, 2023-03-12:
The problem is clearly in your video driver.   Command i2cdetect, which is a standard system tool from package i2-tools, shows that slave address x37 (DDC) is not responsive, while slave address x50 (EDID) is. Normally this would indicate a monitor issue, but we know that your monitor supports DDC, both because of the Lunar test and because its a recent name brand monitor.  So the only thing that's left between i2cdetect and the physical monitor is the video driver.

I can't say much more about the driver, since it isn't using /sys in the usual way, but I suspect it comes from the board manufacturer.

pharris2411, 2023-03-13:
Yeah I figured that would be the case. Thank you for confirming, I'll close this issue but leave it up in case anyone else stumbles across it on Google.

Also kudos for this project -- this is one of the best documented projects I've ever seen!

closed 2023-03-13 by pharris2411