rockowitz/ddcutil #303: DDC communication fails more often than succeeds, and setting features never succeeds.

kennyrkun, 2023-03-19:
Hello, I am trying to control the brightness on a Planar PCT2435 monitor connected to a Raspberry Pi 4. The monitor *should* support DDC, per page 16 of [the user manual](https://www.planar.com/media/441902/planar-pct2435_user-guide.pdf).

Occasionally, `ddcutil detect` reports Display 1 with the correct information about the monitor and `VCP Version: 2.1`. Typically, this command only works one time and using it again after will report "Invalid display", as well as "VCP version: Detection failed". These does not appear to be exclusive to one another, as sometimes it will report "Invalid display" with a correct VCP version, and other times it will report "Display 1" with DDC communication failed.

I will attach the output of `ddcutil interrogate`, because I'm really not sure what's going on :( I believe this issue could be related to #288, "ddcutil sometimes work, sometimes detection error". I created a separate issue because we are using different hardware.

[ddcutil interrogate.txt](https://github.com/rockowitz/ddcutil/files/11010214/ddcutil.interrogate.txt)

rockowitz, 2023-03-19:
First, thank you for creating a separate issue, and also for including the interrogate output. 

You are using a very old release of **ddcutil**, 0.9.2 from November 2018. In particular, it lacks the ***--sleep-multiplier*** option added in **ddcutil** 0.9.8.

Your symptoms (sometimes things work, sometimes they don't) suggest a marginal DDC implementation in the monitor.  The one way that **ddcutil** can sometimes address this problem is by increasing the amount of time from that specified for waiting between sending a DDC request packet and reading the reply packet.  The way to do that is the ***--sleep-multiplier** option, e.g. ***--sleep-multiplier 2.0***.

kennyrkun, 2023-03-19:
Okay, the package is outdated. I have installed **ddcutil** using apt, and I'm not super familiar with apt. Is there a newer version of the package that I can upgrade to, or do I need to compile and install from source? If so, I can compile but I do not know how to install the package manually.

rockowitz, 2023-03-21:
Have you tried using option ***--sleep-multiplier 2.0***?

kennyrkun, 2023-04-02:
It appears the issue persists even when using `--sleep-multiplier 2.0`. Occasionally the command will work, but usually it says `Display not found.`.

[ddcutil interrogate.txt](https://github.com/rockowitz/ddcutil/files/11131051/ddcutil.interrogate.txt)

rockowitz, 2023-04-03:
I'm afraid I don't have any further suggestions.  The statistics in the interrogate output (which you can see if you use the ***--stats*** option on a command) show large numbers of data errors, and even those operations that ultimately succeed often require a large number of retries.  In addition to ***--sleep-multiplier***, you could also try raising the ***--maxtries*** values, but given the extent of data errors I doubt that would resolve the problem.  FYI, the option ***--ddc*** will report the individual data errors.

kennyrkun, 2025-03-12:
Closing as this is likely an issue with my setup, and I am not pursuing the project which used ddcutil anymore.

ddcutil is a great library, my own issues aside :)

closed 2025-03-12 by kennyrkun