rockowitz/ddcutil #306: DDC communication failed (Operation not supported) over HDMI

tao-song-tfs, 2023-03-20:
Hi,

     We use yocto with iMX8 and I added ddcutil to the yocto build. I tried to use it to adjust the brightness of the display.  I got an error when I ran ddcutil detect:

    root@imx8mp-var-dart:~# ddcutil detect
    Invalid display
    I2C bus:  /dev/i2c-6
    EDID synopsis:
       Mfg id:               MST
       Model:                MStar Demo
       Product code:         48
       Serial number:
       Binary serial number: 1 (0x00000001)
       Manufacture year:     2010,  Week: 10
    DDC communication failed
  
   I found a similar issue "DDC communication failed over HDMI #148" but I got a different error message "Operation not supported" when I ran ddcutil interrogate command. Please see the attached file.
   
    Could you give some suggestion of how to resolve the issue?
    
    Thanks!
   
     Regards,
 
     Tom
[ddcutil interrogate output.txt](https://github.com/rockowitz/ddcutil/files/11021000/ddcutil.interrogate.output.txt)

rockowitz, 2023-04-03:
Looking at file  drivers/gpu/drm/bridge/synopsys/dw-hdmi.c in the Linux 6.2.9 source, I see the following at line 443 in function dw_hdmi_i2c_xfer(): 
~~~
 443     if (addr == DDC_CI_ADDR)
 444         /*
 445          * The internal I2C controller does not support the multi-byte
 446          * read and write operations needed for DDC/CI.
 447          * TOFIX: Blacklist the DDC/CI address until we filter out
 448          * unsupported I2C operations.
 449          */
 450         return -EOPNOTSUPP;
 451 
~~~

So it would appear that the Designware HDMI driver does not support DDC.  

This very old [google hit](https://patchwork.kernel.org/project/dri-devel/patch/1443470762-17937-1-git-send-email-vladimir_zapolskiy@mentor.com/) suggests that there was as some point work to enable DDC support.

I suggest you follow up with the folks responsible for the Designware HDMI driver.  If you find out anything more, please post that information here so that others can benefit.

tao-song-tfs, 2023-05-02:
Hi,

     Thanks for looking into this issue. Sorry I didn't get notice when the issue was updated.

     I will give it a try as you suggested and update the issue if I can find a solution.

     Thanks for help!

     Regards,

     Tom