rockowitz/ddcutil #307: DDC communication with ARZOPA portable monitor fails

kalbfuss, 2023-04-01:
The DDC communication with the [ARZOPA portable 17,3" A1 MAX monitor fails](https://www.arzopa.com/products/portable-monitor-17-3-100-srgb-1080p-fhd-hdr-ips-laptop-computer-display).
Admittedly, I am not sure whether the monitor supports DDC/CI. However, the monitor seems to communicate with ddcutil to a certain extent. The "detect" command returns the following output:
```
Invalid display
   I2C bus:  /dev/i2c-1
      DRM connector:                      card0-HDMI-A-1
      Driver:                             Unknown
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-1/name     DesignWare HDMI
      PCI device path:                    /sys/devices/platform/soc/ff600000.bus/ff600000.hdmi-tx/i2c-1
   EDID synopsis:
      Mfg id:               GWD - UNK
      Model:                ARZOPA
      Product code:         371  (0x0173)
      Serial number:        000000000000
      Binary serial number: 1898184704 (0x71240000)
      Manufacture year:     2022,  Week: 51
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.640, 0.330
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 1e e4 73 01 00 00 24 71   ..........s...$q
      +0010   33 20 01 03 80 26 16 78 3a ee 95 a3 54 4c 99 26   3 ...&.x:...TL.&
      +0020   0f 50 54 21 08 00 d1 c0 71 40 81 80 95 00 90 40   .PT!....q@.....@
      +0030   a9 c0 81 40 81 c0 02 3a 80 18 71 38 2d 40 58 2c   ...@...:..q8-@X,
      +0040   25 00 7d d6 10 00 00 1e 00 00 00 fc 00 41 52 5a   %.}..........ARZ
      +0050   4f 50 41 0a 20 20 20 20 20 20 00 00 00 ff 00 30   OPA.      .....0
      +0060   30 30 30 30 30 30 30 30 30 30 30 0a 00 00 00 fd   00000000000.....
      +0070   00 2f 4b 18 54 13 01 0a 20 20 20 20 20 20 01 10   ./K.T...      ..
   DDC communication failed
```
Output of ddcutil interrogate:
[interrogate.txt](https://github.com/rockowitz/ddcutil/files/11129509/interrogate.txt)

rockowitz, 2023-04-03:
See my response to issue #307. As far as I can tell, the Designware HDMI driver does not support DDC/CI. 

Also, from what little documentation I've found on the monitor, it's unclear whether the monitor itself supports DDC/CI.

kalbfuss, 2023-04-07:
Thanks for checking. Do I understand correctly that exposure of an I2C device by the HDMI driver for communication with the monitor is not sufficient, but the HDMI driver must also implement DDC/CI?

rockowitz, 2023-04-07:
Correct.  More precisely, the driver must implement the I2C services required for the DDC/CI protocol, which are used by the code using the driver.

kalbfuss, 2023-04-15:
Thanks for clarifying and all your efforts. It is a pity, but cannot be changed.  I have the monitor connected to a Radxa Zero single-board computer, which I use for a digital photo frame project. I very much like the Radxa Zero, but the software support is incomplete.