rockowitz/ddcutil #308: error ‘errno’ undeclared, 1.5.0-dev build

johndoe71rus, 2023-04-13:
try to build dev version
```
~/build/ddcutil/ddcutil$ ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I config -I m4 --force -I m4 ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'config'.
libtoolize: copying file 'config/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --include=config --include=m4 --force
autoreconf: running: /usr/bin/autoheader --include=config --include=m4 --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:113: installing 'config/compile'
configure.ac:105: installing 'config/missing'
src/app_ddcutil/Makefile.am: installing 'config/depcomp'
autoreconf: Leaving directory `.'
configure: debug messages disabled
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3                                                                                                     
checking for ar... ar                                                                                                                        
checking the archiver (ar) interface... ar                                                                                                   
checking for pkg-config... /usr/bin/pkg-config                                                                                               
checking pkg-config is at least version 0.9.0... yes                                                                                         
checking for gcc option to accept ISO C99... none needed                                                                                     
checking build system type... x86_64-pc-linux-gnu                                                                                            
checking host system type... x86_64-pc-linux-gnu                                                                                             
checking how to print strings... printf                                                                                                      
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
[... 512 lines cut ...]
dsa2.c: In function ‘dsa2_erase_persistent_stats’:
dsa2.c:953:13: warning: implicit declaration of function ‘unlink’ [-Wimplicit-function-declaration]
  953 |    int rc = unlink(stats_fn);
      |             ^~~~~~
dsa2.c:954:18: error: ‘errno’ undeclared (first use in this function)
  954 |    if (rc < 0 && errno != ENOENT)
      |                  ^~~~~
dsa2.c:954:18: note: ‘errno’ is defined in header ‘<errno.h>’; did you forget to ‘#include <errno.h>’?
dsa2.c:954:27: error: ‘ENOENT’ undeclared (first use in this function)
  954 |    if (rc < 0 && errno != ENOENT)
      |                           ^~~~~~
dsa2.c: In function ‘dsa2_restore_persistent_stats’:
dsa2.c:1027:19: error: ‘ENOENT’ undeclared (first use in this function)
 1027 |    if (linect == -ENOENT)
      |                   ^~~~~~
dsa2.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-compound-token-split-by-macro’
make[3]: *** [Makefile:535: dsa2.lo] Error 1
make[3]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src/base'
make[2]: *** [Makefile:809: all-recursive] Error 1
make[2]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src'
make[1]: *** [Makefile:541: all-recursive] Error 1
make[1]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil'
make: *** [Makefile:448: all] Error 2
```

rockowitz, 2023-04-13:
I've added the necessary includes to file dsa2.c and replaced the unlink() call with remove()

johndoe71rus, 2023-04-13:
```
....
make[3]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src/i2c'
Making all in usb
make[3]: Entering directory '/home/johndoe/build/ddcutil/ddcutil/src/usb'
  CC       usb_base.lo
  CC       usb_edid.lo
  CC       usb_displays.lo
  CC       usb_vcp.lo
  CCLD     libusb.la
make[3]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src/usb'
Making all in dynvcp
make[3]: Entering directory '/home/johndoe/build/ddcutil/ddcutil/src/dynvcp'
  CC       dyn_feature_set.lo
  CC       dyn_parsed_capabilities.lo
  CC       dyn_feature_codes.lo
  CC       dyn_feature_files.lo
  CCLD     libdynvcp.la
make[3]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src/dynvcp'
Making all in ddc
make[3]: Entering directory '/home/johndoe/build/ddcutil/ddcutil/src/ddc'
  CC       ddc_common_init.lo
ddc_common_init.c: In function ‘init_tracing’:
ddc_common_init.c:106:70: error: ‘EINVAL’ undeclared (first use in this function)
  106 |              emit_init_tracing_error(errinfo_accumulator, __func__, -EINVAL,
      |                                                                      ^~~~~~
ddc_common_init.c:106:70: note: each undeclared identifier is reported only once for each function it appears in
ddc_common_init.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-compound-token-split-by-macro’
make[3]: *** [Makefile:488: ddc_common_init.lo] Error 1
make[3]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src/ddc'
make[2]: *** [Makefile:809: all-recursive] Error 1
make[2]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src'
make[1]: *** [Makefile:541: all-recursive] Error 1
make[1]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil'
make: *** [Makefile:448: all] Error 2
```
almost done

rockowitz, 2023-04-13:
Fixed and pushed to github.  Thank you for your persistence!  It's immensely helpful.

johndoe71rus, 2023-04-13:
You're welcome. I'm tester. My contribution to open source.

rockowitz, 2023-04-14:
That's a huge contribution.   Especially for ddcutil, which deals with an enormous range of hardware, much of which has a very loose conformance to the DDC/CI spec and widely varies in performance.

The file dsa2.c, which needed errno.h, is new code that dynamically adjusts sleep time up (if required) and down (if possible) from that in the DDC/CI spec.  Some monitors require more sleep time between operations, but most can make do with much less, greatly speeding up ddcutil's elapsed time.  It is currently enabled by option ***--dsa2***, though that will change before release if the algorithm proves useful and stable.   Try using it.   And in general I'd appreciate your comments on ddcutil and ddcui.

closed 2023-04-15 by rockowitz