rockowitz/ddcutil #309: Is there a list of manufacturers that actually support DDC?

joaociocca, 2023-04-13:
I've been testing what I can do with ddcutil and other softwares using my two LG displays (34WK650-W and 25UM65), but of course I've been having issues - some already identified in several issues here, some caused by errors in the display's port drivers (like DisplayPort), other simply because the display seemingly doesn't actually support said feature correctly...

But then, I was mentioning and randomly babbling about it on twitter, and LG's profile asked me more about it. I thought maybe I'd hit jackpot and they were actually going to help me... that ended on a DM I received today, where they say they don't give support or disclose information about "internal display protocol" and they they also don't disclose informations about the communications protocol DDC.

So I thought about looking for a list of manufacturers which are more friendly in this subject - but, of course, being a niche subject, if there is such a list I haven't been able to find it so far.  Is there?

I could, of course, start such a list. I'll take some time later to map which functions actually work in my two models... Some stuff I remember, and listed below:

## LG

Officially gives no support and provides no information about DDC, despite listing it as a special feature.

[image]
(DM from LGdoBrasil, LG's official Brazilian profile.)

Their message translates to:

> Good morning. Thank you for waiting for our reply. We have checked your request with our technical department and we do not provide support or disclose information about an internal monitor protocol and the use of third-party software. We do not provide information about the communication protocol (DDC).

### Models

#### 25UM65

[image]
(Captured from https://www.lg.com/us/monitors/lg-25UM65-P-ultrawide-monitor at 13/04/2023 15:05 UTC)

- Brightness
- Sound volume
- Can change input one-way, display stops DDC communication when input is changed

#### 34WK650-W

[image]
(Captured from https://www.lg.com/us/monitors/lg-34WK650-W-ultrawide-monitor at 13/04/2023 15:03 UTC)

- Brightness
- Sound volume
- (-) Input change fails when called from DisplayPort connection

rockowitz, 2023-04-14:
Welcome to the wide world of monitors.  If I'd appreciated how much variability there is I doubt I'd have made ddcutil  an open source project.  LG is particularly problematic and offensive, but offhand I don't believe any of the biggies are much more forthcoming with information. Years ago I came across some public facing documentation re DDC/CI from Benq, which suggests they may be more responsive.

Unfortunately, things are only going to get worse, as the DDC/CI spec has not been updated in years,  and there are more and more things, such as PBP that each manufacturer implements in their own way.  

I trust you've seen page [Notes on Specific Monitors[(http://www.ddcutil.com/monitor_notes/), though it hasn't kept up with all the information people have reported in the issues.

Re determining what functions your monitor supports, **ddcutil probe** tries to read (almost) every feature number, so it's a way of testing all but RO features. It and the output of **ddcutil capabilities --verbose** are a good place to start.

joaociocca, 2023-04-14:
yeah, saw that page. And I did already run both `probe` and `capabilities --verbose`, still there are some false positives there, at least in my case.

Damn, that's some really sad news. And yeah, PbP was one of the functions I've been trying to activate here, spent an afternoon on it and got no results so far...

closed 2023-04-14 by joaociocca