rockowitz/ddcutil #310: cannot write some features on Samsung U32D970

arnfaldur, 2023-04-28:
I am able to change some settings like:
```
   Feature: 12 (Contrast)
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
```
but not 
```
   Feature: 10 (Brightness)
```
and some others that seem less likely to work, like input source, color presets, etc.

Is there anything that can be done in cases like this?

Here is the result of `ddcutil interrogate`: [ddcutil-interrogate.txt](https://github.com/rockowitz/ddcutil/files/11349147/ddcutil-interrogate.txt)

I have another monitor whose brightness control works as expected using ddcutil. The Samsung monitor's brightness can be changed using an OSD but the buttons are failing and the OSD UX is very poor so I'm hoping to be able to control it using a tool like ddcutil.

arnfaldur, 2023-04-28:
Using `--enable-usb` does not have an effect.

rockowitz, 2023-04-29:
I have a similar U32H750 which, despite what the numbering suggests, appears to be a later version of your U32D970.  The display controller type and firmware level (VCP features xc8 and xc9) are identical.  There are complicated relationships among the various features, which I've not fully worked out.  In particular, if I choose Dynamic Contrast as the MagicBright setting in the OSD, Brightness and other settings are grayed out. Perhaps one of your other settings is disabling brightness (feature x10).  You might want to try resetting all features to factory defaults, either in the OSD or by setting a non-zero value in feature x04 (setvcp x04 1).

arnfaldur, 2023-04-30:
Thanks for the suggestion. A factory reset didn't help unfortunately.