rockowitz/ddcutil #311: only the last connected display is detected

ntzb, 2023-05-02:
Hi,
I have two monitors connected using DP and HDMI, and only the last connected one is detected.
if I unplug one, the other will appear.

I ran `sudo ddcutil interrogate --verbose`
[inerrogate.txt](https://github.com/rockowitz/ddcutil/files/11372137/inerrogate.txt)
and `sudo ddcutil detect --verbose --trace ddc --trace i2c`
[detect.txt](https://github.com/rockowitz/ddcutil/files/11372138/detect.txt)

in these logs, the secondary plugged monitor is detected, but it's much less interesting for DDC usage than the first one.

env:
ASUS laptop on Ubuntu 22.10 with kernel 6.2.8, two fairly recent monitors.
p.s. on macos and same connectivity, they are both controllable using MonitorControl.

any help would be appreciated!

rockowitz, 2023-05-03:
Can you describe in detail your hardware configuration?  From what I can tell, your Asus Vivobook has both a Type-C connector and a HDMI connector for video. (Also a USB 3.2 Type A connector, which is not usable for video.)   One monitor is a Hitachi X93W, which is a very old monitor with a DVI connector.  The second is an LG HDR QHD with a HDMI connector.  Laptop model and monitor models?  What connectors on the laptop are being used?   What connectors on the monitors are being used?  How are the connections made? (e.g. using a Type-C to HDMI dongle)  Do you know if any of the dongles/cables are active or passive?  Do any of the dongles/cables have a DP++ symbol stamped? 

i2cdetect reports no /dev/i2c device for the LG monitor.  This indicates a hardware or i915 driver issue.

ntzb, 2023-05-03:
the laptop has a thunderbolt connection, and I'm using a TB docking station.
the X93W monitor is using DVI to HDMI adapter, and the LG monitor is using DisplayPort.
don't see any DP++ symbol on the cables.
the X93W is actually Hyundai X93W, with 1xVGA and 1xDVI (using the DVI with HDMI adapter, this monitor is correctly identifiable and controllable).
the LG is 27BN65Q with 2xHDMI and 1xDP (I tried both HDMI and DP).

the LG monitor is functioning properly (as a monitor), and as mentioned, detectable in ddcutil if I unplug the X93W monitor

rockowitz, 2023-05-04:
What are the brand and model of your docking station?

Please run interrogate again, this time with only the LG monitor plugged into the docking station.

ntzb, 2023-05-04:
the docking is ASUS SimPro Dock 2.
when only the LG monitor is plugged, it is detected and fully controllable using ddcutil.
[here's](https://github.com/rockowitz/ddcutil/files/11394215/inerrogate.txt) the output of `sudo ddcutil interrogate --verbose`

rockowitz, 2023-05-05:
Your problem appears to be the inverse of that described in [this issue](https://gitlab.freedesktop.org/drm/intel/-/issues/3605) that I posted to the drm/intel issue list some time ago.  In that case, a /dev/i2c device is assigned to each of the MST connected monitors in the docking station, and there is also a /dev/i2c device for one of the connected monitors defined for the computer port to which the docking station is connected.  So one of the monitors has 2 different /dev/i2c devices assigned for it.  If you scroll down in the issue discussion, you'll see an explanation of why this happens, and why there's not a clear solution.

In your case, neither MST port in the dock has a /dev/i2c device assigned to it while, as in the above case, one of the monitors has a /dev/i2c device to the computer port. 

I suggest you report your problem as a new issue on the freedesktop.org drm/intel list.  Alternatively, you could report it as an Ubuntu bug - it will be passed on the i915 developers.  Be sure to include the details of your hardware configuration, including brands and models, kernel version, etc.

ntzb, 2023-05-05:
thanks for the analysis.
can you point to where in the logs you see what you mentioned? so I can give something for developers to look at.
also, can you say where between the two options, freedesktop.org or launchpad.net, is it more appropriate to open the bug report at? 

lastly, from what you see, is there a workaround for selecting one controllable monitor? or is it the last monitor connected getting a /dev/i2c device and nothing can be done about that?

rockowitz, 2023-05-08:
It's more a matter of what's not there.  I would expect to see a /dev/i2c device for the port on the laptop,  and two more for the ports on the docking station.   In your case i2cdetect shows slave address x50 (EDID) active for dev/i2c-13 (laptop screen, /sys/bus/i2c/devices/i2c-13/name = AUX A/DDI A/ PHY A)  and slave addresses x50 and x37(DDC) active for /dev/i2c-14 (laptop USB-C port, /sys/bus/i2c/devices/i2c-14/name =  AUX USBC1/DDI TCI1/PHY TC1).  There are no slave addresses detected for /dev/i2c-19 (name /sys/bus/i2c/devices/i2c-19/name=DPMST)  If two monitors a plugged into the docking station, I would expect to see 2 i2c devices named DPMST, which i2cdetect output for neither showing slave address x50 (EDID) and x37 (DDC) active.

To report the problem, first describe your hardware and the fact that only 1 monitor plugged into the dock is recognized.

With both monitors plugged into the docking station, list the i2c device names:
~~~
ls /sys/bus/i2c/devices/i2c*/name | xargs grep ".*"
~~~

Then for i2c-14 and each of the devices with name DPMST, execute: 
~~~
i2cdetect -y N
~~~
where N is the i2c device number.

You can include the fact that you encountered the problem using ddcutil, but including ddcutil output is unnecessary.  i2cdetect is a basic system utility. 

You can report the problem in either place.   Ultimately it will be forwarded to the i915 driver developers.  Let me know what the bug/issue number is so I can track it.

The only workaround is to connect only 1 monitor the docking station, and connect the other monitor to the laptop's HDMI port.

ntzb, 2023-05-11:
reported here https://bugs.launchpad.net/ubuntu/+source/i2c-tools/+bug/2019174
I'll give it some time and if nothing moves will cross post in freedesktop