rockowitz/ddcutil #312: error 1.5.0-dev build

johndoe71rus, 2023-05-10:
```
 ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I config -I m4 --force -I m4 ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'config'.
libtoolize: copying file 'config/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --include=config --include=m4 --force
autoreconf: running: /usr/bin/autoheader --include=config --include=m4 --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:113: installing 'config/compile'
configure.ac:105: installing 'config/missing'
src/app_ddcutil/Makefile.am: installing 'config/depcomp'
autoreconf: Leaving directory `.'
configure: debug messages disabled
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for gcc option to accept ISO C99... none needed
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
[... 447 lines cut ...]
      |                                                       ^~~~~
../../src/base/core.h:278:51: note: in definition of macro ‘DBGMSG’
  278 |           __func__, __LINE__, __FILE__, format, ##__VA_ARGS__)
      |                                                   ^~~~~~~~~~~
ddc_serialize.c:27:1: note: ‘errno’ is defined in header ‘<errno.h>’; did you forget to ‘#include <errno.h>’?
   26 | #include "ddc_serialize.h"
  +++ |+#include <errno.h>
   27 | 
In file included from ddc_serialize.c:18:
ddc_serialize.c:531:55: note: each undeclared identifier is reported only once for each function it appears in
  531 |       DBGMSG("Error writing file %s:%s", fn, strerror(errno));
      |                                                       ^~~~~
../../src/base/core.h:278:51: note: in definition of macro ‘DBGMSG’
  278 |           __func__, __LINE__, __FILE__, format, ##__VA_ARGS__)
      |                                                   ^~~~~~~~~~~
ddc_serialize.c: At top level:
cc1: warning: unrecognized command line option ‘-Wno-compound-token-split-by-macro’
make[3]: *** [Makefile:490: ddc_serialize.lo] Error 1
make[3]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src/ddc'
make[2]: *** [Makefile:812: all-recursive] Error 1
make[2]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil/src'
make[1]: *** [Makefile:541: all-recursive] Error 1
make[1]: Leaving directory '/home/johndoe/build/ddcutil/ddcutil'
make: *** [Makefile:448: all] Error 2
```

rockowitz, 2023-05-10:
I fixed the missing errno.h problem and pushed the change to github.  Thank you.

Do I understand correctly that the first run of .configure did not detect that libjansson-dev was not installed?  Autoconf macro PKG_CHECK_MODULE() looks for file jansson.pc installed by the development package and terminates configure execution if the file is not found.

closed 2023-05-10 by johndoe71rus

johndoe71rus, 2023-05-10:
run  ./autogen.sh was after install  libjansson-dev.
don't forget add  libjansson-dev to depences