rockowitz/ddcutil #316: DDC communication failed for Eizo CS2740

mesonium, 2023-05-25:
DDC communication fails for Eizo CS2740 using DP and an AMD Polaris Card on openSUSE TW unfortunately. Can you judge from the outputs below if there's anything to do about? Thanks! :)

Please see the output of ```ddcutil --interrogate```:  [ddcutil_interrogate.txt](https://github.com/rockowitz/ddcutil/files/11564201/ddcutil_interrogate.txt)

```
ddcutil detect --verbose --ddc
Invalid display
   I2C bus:  /dev/i2c-6
      DRM connector:                      card1-DP-2
      Driver:                             Unknown
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-6/name     AMDGPU DM aux hw bus 1
      PCI device path:                    /sys/devices/pci0000:00/0000:00:03.1/0000:0a:00.0/drm/card1/card1-DP-2/i2c-6
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                CS2740
      Product code:         12548  (0x3104)
      Serial number:        
      Binary serial number: 12345678 (0x01234567)
      Manufacture year:     2020,  Week: 30
      EDID version:         1.4
      Extra descriptor:        
      Video input definition:    0xb5 - Digital Input (DisplayPort), Bit depth: 10
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.658, 0.319
      Green x,y:        0.213, 0.705
      Blue  x,y:        0.148, 0.063
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 15 c3 04 31 10 ab 7f 01   ...........1....
      +0010   24 1e 01 04 b5 3c 22 78 e2 ba 15 a8 51 36 b4 26   $....<"x....Q6.&
      +0020   10 50 54 a1 08 00 a9 40 81 80 81 40 d1 c0 d1 00   .PT....@...@....
      +0030   01 01 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 20   ......M....p>.0 
      +0040   35 00 54 4f 21 00 00 1a ef 51 00 a0 f0 70 19 80   5.TO!....Q...p..
      +0050   30 20 35 00 54 4f 21 00 00 1a 00 00 00 fd 00 17   0 5.TO!.........
      +0060   3d 19 89 36 00 0a 20 20 20 20 20 20 00 00 00 fc   =..6..      ....
      +0070   00 43 53 32 37 34 30 0a 20 20 20 20 20 20 01 2b   .CS2740.      .+
   DDC communication failed
```

rockowitz, 2023-05-26:
I2C slave address x37 (DDC) is not responsive on this monitor.  i2cdetect for DP connections is often incorrect, but I expect that were installed on your system, and the monitor connected to the HDMI or DVI output, i2c detect would report slave address x50 (EDID) as active, but slave address x37 as inactive.

When developing the USB support in ddcutil, I had the use of 2 different Eizo monitors.    One responded to VCP requests over both I2c and USB, the other only on USB.   Perhaps your monitor only uses USB.  By default, support for monitors that use USB for MCCS communication is turned off in ddcutil.  To turn it on, use option --enable-usb.  Note that this option is only useful if the monitor respects the MCCS over USB spec, which the Eizo monitors did.  Command **ddcutil usbenvironment** analyzes the  environment.

rockowitz, 2023-05-26:
@mesonium I had written the bulk of this reply before I saw you had added the usbenvironment output.

I2C slave address x37 (DDC) is not responsive on this monitor. i2cdetect for DP connections is often incorrect, but I expect that were installed on your system, and the monitor connected to the HDMI or DVI output, i2c detect would report slave address x50 (EDID) as active, but slave address x37 as inactive.

When developing the USB support in ddcutil, I had the use of 2 different Eizo monitors. One responded to VCP requests over both I2c and USB, the other only on USB. Perhaps your monitor only uses USB. By default, support for monitors that use USB for MCCS communication is turned off in ddcutil. To turn it on, use option --enable-usb. Note that this option is only useful if the monitor respects the MCCS over USB spec, which the Eizo monitors did. Command **ddcutil usbenvironment** analyzes the environment.

Based on your additional file, I would simply suggest that you try using option ***--enable-usb***, and run as root to avoid permission issues, e.g. **sudo ddcutil  detect --enable-usb**.

mesonium, 2023-05-26:
> Based on your additional file, I would simply suggest that you try using option --enable-usb, and run as root to avoid permission issues, e.g. sudo ddcutil detect --enable-usb.

I had accidentally overseen the USB-B (upstream, on the monitor) to USB-A (downstream, computer) cable [[1]](https://www.eizo.com/products/coloredge/cs2740/). When I disconnect this cable, I don't get any EIZO related output anymore with `ddcutil usbenvironment --verbose` obviously.

Anyway, `sudo ddcutil detect --enable-usb` using the DP and the above mentioned USB-cable yields `DDC communication failed`.

mesonium, 2023-05-26:
This is how it looks like when I connect it via HDMI (and have the USB cable still connected):

```
sudo ddcutil detect --verbose --ddc (--enable-usb)
Invalid display
   I2C bus:  /dev/i2c-2
      DRM connector:                      card1-HDMI-A-1
      Driver:                             Unknown
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-2/name     AMDGPU DM i2c hw bus 2
      PCI device path:                    /sys/devices/pci0000:00/0000:00:03.1/0000:0a:00.0/i2c-2
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                CS2740
      Product code:         12582  (0x3126)
      Serial number:        
      Binary serial number: 12345678 (0x01234567)
      Manufacture year:     2020,  Week: 36
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.658, 0.319
      Green x,y:        0.213, 0.705
      Blue  x,y:        0.148, 0.063
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 15 c3 26 31 10 ab 7f 01   ..........&1....
      +0010   24 1e 01 03 80 3c 22 78 ea ba 15 a8 51 36 b4 26   $....<"x....Q6.&
      +0020   10 50 54 a1 08 00 a9 40 81 80 81 40 d1 c0 01 01   .PT....@...@....
      +0030   01 01 01 01 01 01 08 e8 00 30 f2 70 5a 80 b0 58   .........0.pZ..X
      +0040   8a 00 54 4f 21 00 00 1e 04 74 00 30 f2 70 5a 80   ..TO!....t.0.pZ.
      +0050   b0 58 8a 00 54 4f 21 00 00 1e 00 00 00 fd 00 17   .X..TO!.........
      +0060   3d 0f 87 3c 00 0a 20 20 20 20 20 20 00 00 00 fc   =..<..      ....
      +0070   00 43 53 32 37 34 30 0a 20 20 20 20 20 20 01 5e   .CS2740.      .^
   DDC communication failed
```

rockowitz, 2023-05-26:
I assume the parentheses around "--enable-usb" in your command line were edited in.  Command **sudo ddcutil detect --verbose --ddc (--enable-usb)** will fail to parse.

Try **sudo ddcutil detect --verbose --enable-usb --trace usb**.  The output will be voluminous, but should help me pinpoint why MCCS over USB doesn't work.  (Option ***--ddc*** is irrelevant here.  It reports invalid DDC/CI data packets, which use I2C.)

mesonium, 2023-05-26:
> I assume the parentheses around "--enable-usb" in your command line were edited in.

Yes, they were supposed to correspond to the sentence above. Sorry for the confusion. ;)

> Try sudo ddcutil detect --verbose --enable-usb --trace usb. The output will be voluminous, but should help me pinpoint why MCCS over USB doesn't work. 

Here's the output, I have just removed non-monitor related lines of other USB devices:

<details>
<summary>sudo ddcutil detect --verbose --enable-usb --trace usb</summary>
<p>

```
(get_usb_monitor_list          ) Examining device: /dev/usb/hiddev2
(is_possible_monitor_by_hiddev_name) Starting.  hiddev_name = /dev/usb/hiddev2
(is_possible_monitor_by_hiddev_name) Before lookup call
(is_possible_monitor_by_hiddev_name) detailed_device_summary: 
      Usb_Detailed_Device_Summary at: 0x55e391c3e000
         devname                                                  : /dev/usb/hiddev2
         vendor_id                                                : 056d
         product_id                                            () : 404d
         vendor_name                                              : EIZO
         product_name                                             : EIZO USB HID Monitor
         busnum_s                                                 : 1
         devnum_s                                                 : 6
         prop_busnum                                              : 001
         prop_devnum                                              : 006
         prop_model                                               : EIZO_USB_HID_Monitor
         prop_model_id                                            : 404d
         prop_usb_interfaces                                      : :030000:
         prop_vendor                                              : EIZO
         prop_vendor_from_database                                : EIZO Corp.
         prop_vendor_id                                           : 056d
         prop_major                                               : 189
         prop_minor                                               : 5
(avoid_device_by_usb_interfaces_property_string) Starting. interfaces = |:030000:|
(avoid_device_by_usb_interfaces_property_string) Returning: false
(is_possible_monitor_by_hiddev_name) Returning: true
(usb_open_hiddev_device        ) hiddev_devname=/dev/usb/hiddev2, calloptions=0xa0 (CALLOPT_ERR_MSG|CALLOPT_RDONLY)
(usb_open_hiddev_device        ) Returning file descriptor: 3
(get_usb_monitor_list          ) open succeeded
(hiddev_get_device_info        ) Starting.
(hiddev_get_device_info        ) Done.     Returning: OK(0): success
Monitor on device /dev/usb/hiddev2 reports no EDID or has invalid EDID. Ignoring.
(get_usb_monitor_list          ) Closed

   GPtrArray of 0 Usb_Monitor_Info at 0x55e391c3b5c0
Invalid display
   I2C bus:  /dev/i2c-6
      DRM connector:                      card1-DP-2
      Driver:                             Unknown
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-6/name     AMDGPU DM aux hw bus 1
      PCI device path:                    /sys/devices/pci0000:00/0000:00:03.1/0000:0a:00.0/drm/card1/card1-DP-2/i2c-6
   EDID synopsis:
      Mfg id:               ENC - Eizo Nanao Corporation
      Model:                CS2740
      Product code:         12548  (0x3104)
      Serial number:        
      Binary serial number: 12345678 (0x01234567)
      Manufacture year:     2020,  Week: 30
      EDID version:         1.4
      Extra descriptor:        
      Video input definition:    0xb5 - Digital Input (DisplayPort), Bit depth: 10
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.658, 0.319
      Green x,y:        0.213, 0.705
      Blue  x,y:        0.148, 0.063
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 15 c3 04 31 10 ab 7f 01   ...........1....
      +0010   24 1e 01 04 b5 3c 22 78 e2 ba 15 a8 51 36 b4 26   $....<"x....Q6.&
      +0020   10 50 54 a1 08 00 a9 40 81 80 81 40 d1 c0 d1 00   .PT....@...@....
      +0030   01 01 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 20   ......M....p>.0 
      +0040   35 00 54 4f 21 00 00 1a ef 51 00 a0 f0 70 19 80   5.TO!....Q...p..
      +0050   30 20 35 00 54 4f 21 00 00 1a 00 00 00 fd 00 17   0 5.TO!.........
      +0060   3d 19 89 36 00 0a 20 20 20 20 20 20 00 00 00 fc   =..6..      ....
      +0070   00 43 53 32 37 34 30 0a 20 20 20 20 20 20 01 2b   .CS2740.      .+
   DDC communication failed

(main                          ) ddcutil execution complete, Fri May 26 15:03:02 2023
```
</p>
</details>

mesonium, 2023-07-07:
I just tested it again with the newest monitor firmware, `ddcutil 2.0.0-rc1` and `sudo ddcutil detect --verbose --enable-usb --trace usb`:

DDC commuication still fails and the diff is quasi the same:

```
< (main                          ) ddcutil execution complete, Fri Jul  7 18:58:33 2023
< (discard_usb_monitor_list      ) Starting  usb_monitors=0x5608a69b71e0, usb_open_errors=0x5608a69d5200
< (discard_usb_monitor_list      ) Done
---
> (main                          ) ddcutil execution complete, Fri Jul  7 18:59:26 2023
```

rockowitz, 2023-07-10:
I'm afraid I don't have anything further to suggest.  The monitor does not appear to adhere to the [USB Monitor Control Class Specification](https://usb.org/sites/default/files/usbmon10.pdf).  To say more I'd need to have that model monitor to test with.