rockowitz/ddcutil #317: DDC communication failed for BenQ BL2405

mesonium, 2023-05-25:
DDC communication fails for BenQ BL2405 using HDMI and an AMD Polaris Card on openSUSE TW unfortunately. Can you judge from the outputs below if there's anything to do about? Thanks! :)

Please see the output of ```ddcutil --interrogate --verbose```:  [ddcutil_interrogate_benq.txt](https://github.com/rockowitz/ddcutil/files/11564268/ddcutil_interrogate_benq.txt)

```
ddcutil detect --verbose --ddc
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
Invalid display
   I2C bus:  /dev/i2c-2
      DRM connector:                      card1-HDMI-A-1
      Driver:                             Unknown
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-2/name     AMDGPU DM i2c hw bus 2
      PCI device path:                    /sys/devices/pci0000:00/0000:00:03.1/0000:0a:00.0/i2c-2
   EDID synopsis:
      Mfg id:               BNQ - UNK
      Model:                BenQ BL2405
      Product code:         32790  (0x8016)
      Serial number:        4CH03595SL0
      Binary serial number: 21573 (0x00005445)
      Manufacture year:     2017,  Week: 50
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.631, 0.351
      Green x,y:        0.334, 0.615
      Blue  x,y:        0.157, 0.051
      Extension blocks: 1
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 09 d1 16 80 45 54 00 00   ............ET..
      +0010   32 1b 01 03 80 35 1e 78 2e ba 45 a1 59 55 9d 28   2....5.x..E.YU.(
      +0020   0d 50 54 a5 6b 80 81 00 81 c0 81 80 a9 c0 b3 00   .PT.k...........
      +0030   d1 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 13 2a 21 00 00 1e 00 00 00 ff 00 34 43 48   E..*!........4CH
      +0050   30 33 35 39 35 53 4c 30 0a 20 00 00 00 fd 00 32   03595SL0. .....2
      +0060   4c 1e 53 11 00 0a 20 20 20 20 20 20 00 00 00 fc   L.S...      ....
      +0070   00 42 65 6e 51 20 42 4c 32 34 30 35 0a 20 01 b2   .BenQ BL2405. ..
   DDC communication failed
```

rockowitz, 2023-05-26:
@mesonium  In contrast to your EIZO monitor, the Benq is responsive on slave address x37.   The problem is that the reply packet is entirely zero.    Try using a large sleep multiplier value, e.g. ***--sleep-multiplier 3***.  Also, see what happens if you put the monitor on the card's DP connector.  In that case I2C communication is "multiplexed" over the more reliable DP-AUX channel.

I myself am using a RX580 card on the the system I am writing from.  The system has 3, sometimes 4, monitors connected, and DDC communication is reliable, so I doubt the problem lies in the card or how AMDGPU supports the card.

mesonium, 2023-05-26:
Thank you for your answers!

> Try using a large sleep multiplier value, e.g. --sleep-multiplier 3

Even ` sudo ddcutil detect --verbose --ddc --sleep-multiplier 20` did not make any difference unfortunately.

>  Also, see what happens if you put the monitor on the card's DP connector.

The monitor only has HDMI, DVI and VGA connectors. :')

rockowitz, 2023-05-26:
I'm afraid I'm out of suggestions.   I assume you've verified that DDC/CI is enabled in the monitor's OSD.

mesonium, 2023-05-26:
> I'm afraid I'm out of suggestions. I assume you've verified that DDC/CI is enabled in the monitor's OSD.

Yes it is enabled in the OSD. Too bad! Thank you anyway for your time and comments!

closed 2023-05-26 by mesonium