rockowitz/ddcutil #318: HP w2207 Invalid Display

joshegrimes, 2023-06-07:
Hello. Computer Engineering student here, new to Raspberry Pi and new to GitHub.

I'm working on a personal project to make a digital picture frame using a Raspberry Pi 4B (4GB) and would like to be able to control the brightness of an HP w2207 monitor using ddcutil (or any method if possible).

Unfortunately, when running "ddcutil detect --verbose" I get the following:

```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false

Open failed for /dev/usb/hiddev0: errno=EACCES(13): Permission denied
  USB bus 1, device 3, vid:pid: 046d:c52b - Logitech:USB Receiver
Invalid display
   I2C bus:             /dev/i2c-21
      I2C address 0x30 (EDID block#)  present: false
      I2C address 0x37 (DDC)          present: true
      I2C address 0x50 (EDID)         present: true
      /sys/bus/i2c/devices/i2c-21/name: Broadcom STB :
   EDID synopsis:
      Mfg id:           HWP
      Model:            HP w2207
      Serial number:    CND731128D
      Manufacture year: 2007
      EDID version:     1.3
      Product code:     9897
      Extra descriptor: Unspecified
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.646, 0.339
      Green x,y:        0.290, 0.603
      Blue  x,y:        0.145, 0.065
      Extension blocks: 0
   EDID source:
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c  
      +0000   00 ff ff ff ff ff ff 00 22 f0 a9 26 01 01 01 01   ........"..&....
      +0010   1f 11 01 03 80 2f 1e 78 ee b5 35 a5 56 4a 9a 25   ...../.x..5.VJ.%
      +0020   10 50 54 a5 6b 90 71 00 81 40 95 00 a9 00 b3 00   .PT.k.q..@......
      +0030   01 01 01 01 01 01 21 39 90 30 62 1a 27 40 68 b0   ......!9.0b.'@h.
      +0040   36 00 d9 28 11 00 00 1c 00 00 00 fd 00 30 4c 18   6..(.........0L.
      +0050   52 10 00 0a 20 20 20 20 20 20 00 00 00 fc 00 48   R...      .....H
      +0060   50 20 77 32 32 30 37 0a 20 20 20 20 00 00 00 ff   P w2207.    ....
      +0070   00 43 4e 44 37 33 31 31 32 38 44 0a 20 20 00 01   .CND731128D.  ..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```

I am using a micro/mini (?) HDMI to DVI cable (i.e. no adapter) from StarTech directly from my Pi to the DVI port. Another note is that yes, I have checked the monitor's "DDC/CI Support" setting and it is enabled.

Was wanting to give up with this monitor, but noticed on ddcutil.com under "Installation and Configuration" and "Notes on Specific Monitors" it states that my monitor (HP w2207) "Responds with data to all VCP feature requests, never reports a feature as unsupported." So that seemed odd given my troubles.

Just posting hoping that someone might have a bit of knowledge to help me out here. Thanks in advance.

rockowitz, 2023-06-08:
First, please submit the output of **sudo ddcutil interrogate** as an attachment.  It shortcuts lots of back and forth with specific questions.

Since you're building from GitHub, please build from the 2.0.0-dev branch.  It has some improvements in reporting display detection errors.

I assume you've looked at the [Raspberry Pi](https://www.ddcutil.com/raspberry) page on the web site that contains configuration information for Pi 4 that was submitted by a user.  I only have a Pi 3 here so have not tested myself.

You say that you're using a micro-HDMI to DVI cable between the Pi and the monitor, but also allude to a StarTech device of some sort.   What is it and how is it used?  Or is it just the brand of HDMI-DVI cable?

Have you tested the Pi with another monitor, preferably one with a HDMI input.   (HDMI-DVI SHOULD work, but it's possible that the cable doesn't connect the I<sup>2</sup>C lines.)

joshegrimes, 2023-06-08:
- Here is the ddcutil interrogate text output:
[ddcutil interrogate.txt](https://github.com/rockowitz/ddcutil/files/11692873/ddcutil.interrogate.txt)

- I don't exactly understand what you mean when you're saying that I'm "building from GitHub" and I don't know how to build from a new branch... sorry for my lack of knowledge here but like I said I'm brand new to Raspberry Pi, Linux, and GitHub.

I do understand that I'm using a way outdated version of ddcutil, but I don't know how to install a newer version of it since I've just been using the "apt" command on Raspbian to install, and when I do that on my version of Raspbian, it installs that 0.9.2 version of ddcutil...

- I have looked extensively at that RPi page on the website and have done all the configuration steps exactly as it is stated. I've also looked at every Raspberry Pi related issue for ddcutil here on this GitHub page but nothing has solved my issue so far.

- Sorry for the confusion with the StarTech thing. StarTech is simply the brand of the cable.

- I have tested the Pi with 2 other monitors and with different inputs on all 3 monitors. So far, with the HP w2207 monitor I've done micro HDMI to DVI and micro HDMI to VGA - and both gave the "Invalid display" message. I've also tried with an ASUS VP247 monitor using micro HDMI to DVI, micro HDMI to VGA, and micro HDMI to HDMI. On this monitor too, I get the "Invalid display" message using any of these connections. Finally, I tried with an ASUS PB277 monitor using micro HDMI to DVI and micro HDMI to HDMI, and this actually **did work** for both types of connections. It gave the "Display 1" message and I was able to change the brightness using setvcp and all that. So, that at least proves that my micro HDMI to DVI cable is not the issue. Of course, though, I ultimately want to be able to use the HP w2207 monitor for this project.

I'm not sure if finding a way to update my version of ddcutil could fix this issue or if it is ultimately just a problem with the monitor, monitor-driver compatibility, or something else. But, I guess I'm trying not to lose hope since ddcutil has supposedly worked with this monitor for other users as I mentioned. Thanks again for your help and time!!

rockowitz, 2023-06-11:
Frankly, I'm at a loss as to what to suggest.  First, that ddcutil works with the PB277 indicates that the problem is not with ddcutil. The error code I'm seeing (EREMOTEIO) in your interrogate output is a hard error within the video driver.   Perhaps there's a problem in your driver configuration, though given that the PB277 works suggests not.

Yet I'm surprised that  DDC  communication works with the PB277 but fails with the VP247. The latter is a fairly recent monitor (2016) as best I can tell, and I would expect it to support DDC/CI.  If you have a desktop Linux system, you could try testing your monitors on that system.

Finally, Raspian builds for the latest ddcutil release (1.4.2) are available on the [OpenSuse Build Service](https://software.opensuse.org/package/ddcutil). 

I'm sorry I don't have more suggestions for you.  You might try posting on one of the Raspian forums re the video driver configuration.

joshegrimes, 2023-06-11:
Thanks for the help with installing the latest release, I did that and it is properly running. That did not fix the issue, unfortunately, even though I know you didn't expect it to.

No worries about having no more suggestions, I've come to realize that this is a very random one-off issue with my desired setup, and I actually already purchased another monitor that is A) better for this project anyway and B) works with ddcutil :). Thanks again for all your help with this, definitely learning a lot with this project and you have been a big part of that. Really appreciate the time you've put in!!