rockowitz/ddcutil #319: Can not set brightness on iiyama ProLite XUB2792HSU

Joku1806, 2023-06-11:
Hello,

When trying to set the brightness on my monitor with `ddcutil setvcp 10 50`, it outputs `Verification failed for feature 10`.
I found [this issue](https://github.com/rockowitz/ddcutil/issues/136), which is similar to mine.

But the solution for this monitor, executing `ddcutil scs` after the brightness command, does not work for me.

I also can't adjust the brightness in the monitor menu, the brightness related entry is greyed out.
I have attached the output of `ddcutil environment --verbose` [here](https://github.com/rockowitz/ddcutil/files/11714397/ddcutil_env.log).
Any idea what could be the problem?

Thank you and have a nice day!

rockowitz, 2023-06-11:
You most likely have the monitor set to a mode that disables the brightness control, such as Cinema display, the ECO brightness adjustment, blue light adjustment, etc.  I suggest resetting the monitor to factory defaults in the OSD.

Joku1806, 2023-06-11:
Yeah that did the trick, thanks!

closed 2023-06-11 by Joku1806