rockowitz/ddcutil #324: Invalid display when running `ddcutil detect`

VictorQueiroz, 2023-06-26:
When I run `ddcutil detect --verbose`, I get the following output:

```
Invalid display
   I2C bus:  /dev/i2c-4
      DRM connector:                      card0-eDP-1
      Driver:                             i915
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      true 
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-4/name     AUX A/DDI A/PHY A
      PCI device path:                    /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/i2c-4
   EDID synopsis:
      Mfg id:               CMN - Chimei Innolux Corporation
      Model:                
      Product code:         5607  (0x15e7)
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2020,  Week: 44
      EDID version:         1.4
      Extra descriptor:        N156HCA-EAB
      Video input definition:    0xa5 - Digital Input (DisplayPort), Bit depth: 8
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.590, 0.350
      Green x,y:        0.330, 0.555
      Blue  x,y:        0.153, 0.119
      Extension blocks: 0
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 0d ae e7 15 00 00 00 00   ................
      +0010   2c 1e 01 04 a5 22 13 78 02 28 65 97 59 54 8e 27   ,....".x.(e.YT.'
      +0020   1e 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 36 36 80 a0 70 38 20 40 30 20   ......66..p8 @0 
      +0040   35 00 58 c1 10 00 00 18 00 00 00 fe 00 4e 31 35   5.X..........N15
      +0050   36 48 43 41 2d 45 41 42 0a 20 00 00 00 fe 00 43   6HCA-EAB. .....C
      +0060   4d 4e 0a 20 20 20 20 20 20 20 20 20 00 00 00 fe   MN.         ....
      +0070   00 4e 31 35 36 48 43 41 2d 45 41 42 0a 20 00 4e   .N156HCA-EAB. .N
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.
```

This is the output of `ddcutil interrogate` command:

[interrogate.txt](https://github.com/rockowitz/ddcutil/files/11872081/interrogate.txt)

The actual problem is, `card0-eDP-1` is my laptop display, and I have a monitor connected through HDMI. I am using `optimus-manager` to use the `nvidia` driver. The external monitor does not work without nvidia driver.

The main point is to be able to change the external monitor brightness using ddcutil, but I am unable to. How to fix that?

The external monitor is an old LG monitor, not sure if it's simply not supported.

rockowitz, 2023-06-26:
I'm afraid I don't have  an answer for you.  The Nvidia driver exports buses /dev/i2c-8 and /dev/i2c-9.  ddcutil find an EDID on neither.  Attempting to use DDC results a hard error (EIO).  i2cdetect does not report slave addresses x50 (EDID) and x37 (DDC) as active, so this really is a low level problem.   I do see that xrandr reports the EDID for the LG monitor.  (It's manufacture date is 2015, which is recent in terms of DDC support.)  But xrandr has ways other than I2C to access the EDID, e.g. by BIOS calls.   All I can suggest is to try a different cable, or try using the nouveau open source driver.

digitaltrails, 2023-06-27:
Could this be related to issue #321.