rockowitz/ddcutil #326: No display found on LG32GN600-B

BlackOutedMind, 2023-07-05:
I installed DDCUTIL in may when I got my monitor and it was working fine, after some system updates ddcutil is not finding any monitor.
Currently only using nvidia graphics card (selected on my mux switch to be directly it) and now ddcutil wont recognize my monitor anymore.
I am suspecting as I turned off optimus-manager and switch to nvidia dGPU only created this issue but I have no proof and don't know how to check it.
[ddcutil.txt](https://github.com/rockowitz/ddcutil/files/11953484/ddcutil.txt)
Any tips to make it work again without having to go back to optimus manager?

rockowitz, 2023-07-10:
The following reply was sent by email on 7/4: 

You do not appear to have installed the [special settings](https://www.ddcutil.com/nvidia/) sometimes needed by the Nvidia driver. System utility i2cdetect reports no slave addresses responsive on any of the Nvidia /dev/i2c devices.

BlackOutedMind, 2023-08-11:
After updating my system it got working again, the latest updates solved some issues I had recently like getting stuck in the process of changing. For example it, when going from 100 to 10 it stuck at 40-ish, so i had to do it again.

closed 2023-08-11 by BlackOutedMind